xArm-Developer/xarm_ros #3: /usr/bin/ld: cannot find -lxarm_ros_client

gcullie, 2019-03-12:
ROS: Melodic
Ubuntu 18.04
Gazebo: 9

Errors     << xarm_controller:make /home/ubuntu/workspace/catkin-ws/logs/xarm_controller/build.make.001.log                                                       
/usr/bin/ld: cannot find -lxarm_ros_client
collect2: error: ld returned 1 exit status
make[2]: *** [/home/ubuntu/workspace/catkin-ws/devel/.private/xarm_controller/lib/xarm_controller/xarm_traj_controller] Error 1
make[1]: *** [CMakeFiles/xarm_traj_controller.dir/all] Error 2
make: *** [all] Error 2
cd /home/ubuntu/workspace/catkin-ws/build/xarm_controller; catkin build --get-env xarm_controller | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

penglongxiang, 2019-03-13:
Dear @gcullie, it looks like 'xarm_ros_client' library is not successfully built. It depends on 'xarm_msgs' header files. Recently we discovered that sometimes those headers fail to get generated before compiling 'xarm_ros_client' during catkin_make. 

You can try to edit xarm_api/CMakeLists.txt by adding:
```bash
add_dependencies(xarm_ros_client xarm_msgs_generate_messages_cpp) 
```
after the line: 
add_library(xarm_ros_client
  src/xarm_ros_client.cpp
)
Thus to make sure the packages are built in proper order. Refer to [this page](https://answers.ros.org/question/188982/message-headers-wont-build-first/). Please let us know if this issue can be solved this way, thank you.

gcullie, 2019-03-14:
I made the suggested changes but the error still exists.

penglongxiang, 2019-03-15:
Could you please provide more information, like every line printed out when executing catkin_make? Thank you.

MichalXh, 2019-06-21:
Hi, I've got the same issue.

Ubuntu 16.04 + ROS kinectic
using catkin_tools

1st I'm getting
```
Errors     << xarm_controller:cmake /home/michal/workspace/ros/logs/xarm_controller/build.cmake.000.log                                                                                                           
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "xarm_api" with any
  of the following names:

    xarm_apiConfig.cmake
    xarm_api-config.cmake

  Add the installation prefix of "xarm_api" to CMAKE_PREFIX_PATH or set
  "xarm_api_DIR" to a directory containing one of the above files.  If
  "xarm_api" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:10 (find_package)
```

Which can be either resolved by editing `CMakeLists.txt` or calling `catkin build xarm_api`, I'm lazy - I choose the latter.

Then I build it again and getting
```
Errors     << xarm_controller:make /home/michal/workspace/ros/logs/xarm_controller/build.make.000.log                                                                                                             
/usr/bin/ld: cannot find -lxarm_ros_client
collect2: error: ld returned 1 exit status
make[2]: *** [/home/michal/workspace/ros/devel/.private/xarm_controller/lib/xarm_controller/xarm_traj_controller] Error 1
make[1]: *** [CMakeFiles/xarm_traj_controller.dir/all] Error 2
make: *** [all] Error 2
```

Which I do not know how to resolve.

building with `catkin_make` works fine.