xArm-Developer/xarm_ros #5: compile errors building xarm_ros

asantanna-applied, 2019-05-29:
Hi  Amy,

Another technical question ....

In attempting to start up Gazebo as per the ReadMe.md file, I have problems
with step 4.3.

## 4.3 Build the code
   ```bash
   $ cd ~/catkin_ws
   $ catkin_make
   ```

ROS runs OK,  Gazebo starts up fine.  But running the make
generates the following errors.   Any ideas?

Thanks!

Andre

============ ERRORS ===========================

test@ubuntu:~/catkin_ws$ catkin_make
In file included from /home/test/catkin_ws/src/ros_control/combined_robot_hw/include/combined_robot_hw/combined_robot_hw.h:37:0,
                 from /home/test/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:29:
/home/test/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/robot_hw.h:168:14: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
 typedef std::shared_ptr<RobotHW> RobotHWSharedPtr;
              ^
In file included from /home/test/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:29:0:
/home/test/catkin_ws/src/ros_control/combined_robot_hw/include/combined_robot_hw/combined_robot_hw.h:107:15: error: ‘RobotHWSharedPtr’ is not a member of ‘hardware_interface’
   std::vector<hardware_interface::RobotHWSharedPtr> robot_hw_list_;
               ^
/home/test/catkin_ws/src/ros_control/combined_robot_hw/include/combined_robot_hw/combined_robot_hw.h:107:15: error: ‘RobotHWSharedPtr’ is not a member of ‘hardware_interface’
/home/test/catkin_ws/src/ros_control/combined_robot_hw/include/combined_robot_hw/combined_robot_hw.h:107:51: error: template argument 1 is invalid
   std::vector<hardware_interface::RobotHWSharedPtr> robot_hw_list_;
                                                   ^
/home/test/catkin_ws/src/ros_control/combined_robot_hw/include/combined_robot_hw/combined_robot_hw.h:107:51: error: template argument 2 is invalid
/home/test/catkin_ws/src/ros_control/combined_robot_hw/include/combined_robot_hw/combined_robot_hw.h:116:49: error: ‘hardware_interface::RobotHWSharedPtr’ has not been declared
                             hardware_interface::RobotHWSharedPtr robot_hw);
                                                 ^
/home/test/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp: In member function ‘virtual bool combined_robot_hw::CombinedRobotHW::prepareSwitch(const std::__cxx11::list<hardware_interface::ControllerInfo>&, const std::__cxx11::list<hardware_interface::ControllerInfo>&)’:
/home/test/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:65:17: error: ‘RobotHWSharedPtr’ is not a member of ‘hardware_interface’
     std::vector<hardware_interface::RobotHWSharedPtr>::iterator robot_hw;
                 ^
/home/test/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:65:17: error: ‘RobotHWSharedPtr’ is not a member of ‘hardware_interface’
/home/test/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:65:53: error: template argument 1 is invalid
     std::vector<hardware_interface::RobotHWSharedPtr>::iterator robot_hw;
                                                     ^
/home/test/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:65:53: error: template argument 2 is invalid
/home/test/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:65:65: error: expected initializer before ‘robot_hw’
     std::vector<hardware_interface::RobotHWSharedPtr>::iterator robot_hw;
                                                                 ^
/home/test/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:66:10: error: ‘robot_hw’ was not declared in this scope
     for (robot_hw = robot_hw_list_.begin(); robot_hw != robot_hw_list_.end(); +

penglongxiang, 2019-05-29:
Dear @asantanna-applied , it seems you have download the **ros_control package** from source, and the compilation errors comes from this source code but not xArm packages. 

We suggest you can try google what are the cause of those errors, or you can just remove this source package from catkin workspace and install ros_control and ros controllers from debian package by:
```bash
sudo apt-get install ros-kinetic-ros-control ros-kinetic-ros-controllers
```

asantanna-applied, 2019-05-29:
Thanks Peng(?),

You are correct.  I just removed the source and everything seems to work
just fine.  Both RViz and Gazebo start up with no problems.

closed 2019-05-29 by asantanna-applied