xArm-Developer/xarm_ros #12: xArm7 gripper

FabianPG11, 2019-10-18:
Currently I am working with the xArm7 robot and I'm trying to control the gripper through moveit/Rviz but I can not manage to load the gripper controller. I can plan and execute a new pose for the manipulator but not for the gripper.

Following the xarm6_gripper_moveit_config package, the controllers.yaml and ros_controllers.yaml files just contain definition for the links 1 to 6, but there is no control for the drive_joint. It is just defined on the hardware_interface. 

my control.yaml looks like:
```

controller_list:
#  - name: xarm7_controller
  - name: ""
    action_ns: follow_joint_trajectory
    default: True
    type: FollowJointTrajectory
    joints:
      - joint1
      - joint2
      - joint3
      - joint4
      - joint5
      - joint6
      - joint7
  - name: end_effector_controller
    action_ns: follow_joint_trajectory
    default: True
    type: FollowJointTrajectory
    joints:
      drive_joint
```

[ INFO] [1571396632.556916925]: Returned 1 controllers in list
--
Can it be related to the fact that the - name: "" in the yaml file must be empty?, because if I put any value the robot it can not find any controller.

To sum up, how can I move the gripper using moveit! and Rviz and get it connected to the real robot?

penglongxiang, 2019-10-21:
Dear @FabianPG11 , sorry the xArm Gripper support for direct Moveit control is not ready yet. Currently, you can use the provided ROS services stated in readme section "[5.7.7 Gripper Control](https://github.com/xArm-Developer/xarm_ros#gripper-control)" to control the Gripper.

pfaraone, 2020-01-22:
@penglongxiang When do you anticipate it will be ready?

mmatsi, 2020-08-20:
@penglongxiang  Almost 1 year passed, are there any updates?

penglongxiang, 2020-08-20:
please refer to: https://github.com/xArm-Developer/xarm_ros/issues/35#issuecomment-677520003