xArm-Developer/xarm_ros #13: XArm questions on Gazebo

eibow1012, 2019-11-04:
Start Gazebo with the following command.
roslaunch xarm_gazebo xarm6_beside_table.launch add_gripper: = true
At that time, there are two questions.
1)
It seems that the gripper is not displayed properly. What is the cause?
2)
Which service should I use when trying to run XArm on Gazebo?
The list of current services is as follows.
$ rosservice list
/ gazebo / apply_body_wrench
/ gazebo / apply_joint_effort
/ gazebo / clear_body_wrenches
/ gazebo / clear_joint_forces
/ gazebo / delete_light
/ gazebo / delete_model
/ gazebo / get_joint_properties
/ gazebo / get_light_properties
/ gazebo / get_link_properties
/ gazebo / get_link_state
/ gazebo / get_loggers
/ gazebo / get_model_properties
/ gazebo / get_model_state
/ gazebo / get_physics_properties
/ gazebo / get_world_properties
/ gazebo / pause_physics
/ gazebo / reset_simulation
/ gazebo / reset_world
/ gazebo / set_joint_properties
/ gazebo / set_light_properties
/ gazebo / set_link_properties
/ gazebo / set_link_state
/ gazebo / set_logger_level
/ gazebo / set_model_configuration
/ gazebo / set_model_state
/ gazebo / set_parameters
/ gazebo / set_physics_properties
/ gazebo / spawn_sdf_model
/ gazebo / spawn_urdf_model
/ gazebo / unpause_physics
/ gazebo_gui / get_loggers
/ gazebo_gui / set_logger_level
/ gazebo_gui / set_parameters
/ robot_state_publisher / get_loggers
/ robot_state_publisher / set_logger_level
/ rosout / get_loggers
/ rosout / set_logger_level
/ xarm / controller_manager / list_controller_types
/ xarm / controller_manager / list_controllers
/ xarm / controller_manager / load_controller
/ xarm / controller_manager / reload_controller_libraries
/ xarm / controller_manager / switch_controller
/ xarm / controller_manager / unload_controller
/ xarm / xarm6_traj_controller / query_state

that's all, thank you very much.

penglongxiang, 2019-11-04:
@eibow1012 the xArm Gripper is currently not fully supported under Gazebo and Moveit, the cause may be there is no effective controller loaded for the gripper, we will be working on this issue, meanwhile you can try to add one by yourself and give proper pid parameters. After launching the model in gazebo, you can run "roslaunch xarm6_gripper_moveit_config xarm6_gripper_moveit_gazebo.launch" to command the virtual robot with Moveit Interface, but the gripper action (open/close) is not supported now.

Or if you are OK with simulation just under RViz without Gazebo, you can just run "roslaunch xarm6_gripper_moveit_config demo.launch" and there will be no problem displaying the gripper since there is no Physics such as gravity simulated. You can drag the marker attached to the tool and execute motion plans in the GUI as well.

eibow1012, 2019-11-05:
Thank you very much.
I will try it as instructed.

ilidar, 2020-01-02:
@penglongxiang Hi, faced the same issue that gripper is not supported yet. In meanwhile [these](https://github.com/StanleyInnovation/robotiq_85_gripper) guys have similar gripper that is working in gazebo. They're using mimic plugin for gazebo from [here](https://github.com/roboticsgroup/roboticsgroup_gazebo_plugins).