xArm-Developer/xarm_ros #14: xArm API to control simulation

anonymouze, 2019-11-10:
While we wait for my xArm to ship, I'd like to start prototyping the code to control it.

Can we use the xarm_ros and xArm-Python-SDK together to achieve it ? So we can write code with SDK functions, and observe motion in Gazebo.

penglongxiang, 2019-11-11:
@anonymouze  actually in xarm_ros you will control the xArm under Moveit! planning algorithm, which is not the same as the one used by our controller (called by xArm python API). So I guess you can simulate in ROS for reference, and get a feeling about the operational space. However if your final project will be implemented by xArm-Python-SDK, the execution may not be exactly the same with Moveit Planner results.

anonymouze, 2019-11-11:
@penglongxiang , thanks. 
Will moveit! planning be roughly similar to what a controller performs ? Also, is there a way to feed data to moveit! with the same or similar interface as is used with the SDK (either python or C++) ?