xArm-Developer/xarm_ros #16: manual control using ps4 joystick

han-droid, 2020-01-02:
Hello,

I'm trying to make a node to control xarm7 using ps4 joystick without moveit.
Currently xarm7 has only position mode, not velocity mode.  Am I right?  
So it is hard to implement joystick manual control.

Is there a way to implement joystick manual control ?

penglongxiang, 2020-01-04:
We do not have a velocity mode yet, but this mode is already under our assessment. I suppose you want to control Cartesian velocity instead of joint velocity right? Currently you may first give xArm a far Cartesian destination toward desired direction with specified speed, and then set mode to be 4 (stop mode) if you want it to stop.

Msani4228, 2020-04-01:
Hello,
I am trying to make a node to control xarm7 using ps4 joystick , without moveit and also target tracking without moveit by only using ROS xArm_Api/xArm_msgs.
(1) it is hard to implement joystick manual control.
 Is there a way to implement joystick manual control ?

(2)Like the way  @penglongxiang said the velocity mode is already under assessment, I wanna ask ,is it already available?

 (3) @penglongxiang  said : << first give xArm a far Cartesian destination toward desired direction with specified speed, and then set mode to be 4 (stop mode) if you want it to stop.>> by using this technique how to implement the Joystick manual control?

(4) My goal is: I want to use velocity controller or position controller with the Joystick or Target tracking. For example giving a target pose [A], before reaching the goal pose [A], we give it another new target pose[B],is there a way to cancel or deviate the previous target [A] and accomplish the new target [B]? if yes how to implement it?
THANKS!!

penglongxiang, 2020-04-03:
Hi, @Msani4228 
We now have a new position control function that allows continuous feeding of Cartesian targets, it is called "servo_cartesian", please check the instructions at https://github.com/xArm-Developer/xarm_ros/tree/master/examples#2-servo_cartesian-streamed-cartesian-trajectory

Currently, you can achieve velocity control by calling the "/xarm/move_servo_cart" at a fixed frequency, and updating the target coordinate incrementally toward the direction you want. If use a fixed frequency of 100Hz, the velocity will be determined by the incremental length you give each time. 

Please use this service with special care, by not giving too large step at high update rate, and currently(firmware v1.4.1), this mode is not monitoring the joint angle limits, this issue will be fixed in the next firmware update within April 2020.

Msani4228, 2020-04-21:
ok Thank you so much.
it worked

closed 2020-04-22 by penglongxiang

Msani4228, 2020-04-22:
Hello !!
For the ServoCartersian control no problem now. 
but for the joint control of the End Effector, the rotation seems not to be working. if we try to rotate the TCP around around the z axis, it moves in Cartesiantrajectory instead of rotating.
so how to rotate the TCP around the Z axis without rotating the xArm Base.
Thanks!

penglongxiang, 2020-04-22:
Hi @Msani4228,

Sorry I am not quite sure about the issue. Could you provide more detailed information for us to reproduce and diagnose? Like which command and input you gave for execution, and the initial position of the arm. Is the model xArm6 you are using?

Msani4228, 2020-04-22:
we are using xArm 7 with Joystick ps4 with xArm Api/msg in ServoCartesian mode.
However when we try to rotate the TCP end Effector in the TCP coordinates base , the TCP moves in line instead of rotating.
we give it input of Yaw so to rotate the TCP end Effector around Z axis.
The main goal is we want the TCP(Joint 7) rotates around its own coordinates base instead of moving in line.

Msani4228, 2020-04-22:
we are using xArm 7 with Joystick ps4 with xArm Api/msg in ServoCartesian mode.
However when we try to rotate the TCP end Effector in the TCP coordinates base , the TCP moves in line instead of rotating.
we give it input of Yaw so to rotate the TCP end Effector around Z axis.
The main goal is we want the TCP(Joint 7) rotates around Z axis in its own coordinates which is the TCP coordinates instead of rotating in base Coordinates .

Sent from my iPhone7

> On Apr 22, 2020, at 13:37, Abdou Yahouza Mahamane Sani <mahamanesania@ymail.com> wrote:
> 
> we are using xArm 7 with Joystick ps4 with xArm Api/msg in ServoCartesian mode.
> However when we try to rotate the TCP end Effector in the TCP coordinates base , the TCP moves in line instead of rotating.
> we give it input of Yaw so to rotate the TCP end Effector around Z axis.
> The main goal is we want the TCP(Joint 7) rotates around its own coordinates base instead of moving in line.

penglongxiang, 2020-04-22:
Hi @Msani4228 ， So you would like to use servoCartesian interface to command the robot to move in Tool Coordinate instead of Base coordinate right? This function will be supported in the next update hopefully in early May.

Msani4228, 2020-04-22:
Yes exactly. So Basically saying for now  if we want to rotate the joint 7 the TCP in its own coordinates while we are in ServoCartesian mode, we definitely need to switch to the Move_Joint mode, right?

penglongxiang, 2020-04-22:
@Msani4228 There is also a similar service called "**/xarm/move_servoj**", you can use this service to control joint motion in the same mode with servoCartesian. It receives **absolute** joint positions as command.  Before calling it, please check the current joint position in "/xarm/joint_states" topic, and increase the target joint position **little by little** just like calling /xarm/move_servo_cart.

For example, if /xarm/joint_state says current joint positions are:  [0.25,-0.47,0.0,-0.28,0.0,0.76,0.24], you can call:  
```bash
rosservice call /xarm/move_servoj [0.25,-0.47,0.0,-0.28,0.0,0.76,0.25] 0.0 0.0 0.0 0.0
```
Which will move joint7 by 0.01 rad **immediately**. Keep calling it and increase the joint positions a small step each time, it will move smoothly. **Be careful not to give a target too far away in one single update**.

Msani4228, 2020-04-22:
ok thank you so much!