xArm-Developer/xarm_ros #17: unnormalized quaternions

bmbyzy, 2020-04-08:
When roslaunch xarm*_moveit_config demo.launch, there are warning about unnormalized quaternions in that last link, which cause rviz to crash during development. How to fix this?

`[ WARN] [1586333539.548156725]: Interactive marker 'EE:goal_link6' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.`

penglongxiang, 2020-04-11:
Hi,

May I ask what is the version of rviz you are using? When running on my computer, the "unnormalised quaternion" message is also printed out, but it would not cause rviz to crush and the marker can be visualised and dragged normally. I am using ros-kinetic-rviz version 1.12.17.  

You can check by running:
```bash
dpkg -l | grep rviz
```

bmbyzy, 2020-04-16:
Hi, thank you for your reply. The crash is caused by another issue in the development. I'm closing this issue now.

closed 2020-04-16 by bmbyzy