xArm-Developer/xarm_ros #18: Gripper in gazebo simulation

nkise, 2020-05-06:
Not propper physical behaviour of gripper in Gazebo simulation. I have used this official (https://github.com/xArm-Developer/xarm_ros/tree/master/xarm_gripper) repository and run xarm6 with gripper. As you can see at this video (https://we.tl/t-i6aSCR2tpx) gripper physic doesn't propperly working. Please can you show how to fix it

Thank You

penglongxiang, 2020-05-13:
Hi @nkise ,

The support of Gripper is not very good for now. We will try to improve it in the following days. Sorry about the inconvenience.

nkise, 2020-05-13:
Hi @penglongxiang , thanks for the response. Are there any terms how long it took to fix the issue?

penglongxiang, 2020-05-14:
@nkise it is hard to say now. xArm gripper is implemented by "mimic joints" which seems not well supported in Gazebo. We have to look into it and see if there is a way around.

nkise, 2020-05-14:
@penglongxiang it seems that on of fixes can be based on this plugin - https://github.com/roboticsgroup/roboticsgroup_gazebo_plugins

I have such result by using it
[image]

nkise, 2020-05-14:
[image]

penglongxiang, 2020-05-18:
@nkise with this mimic_joint_plugin I managed to make the parts hold in stationary state in Gazebo, however I am not sure yet the PID parameters I gave are suitable for motion command. You can pull the latest code and try with:   
```bash
$ roslaunch xarm_gazebo xarm6_beside_table.launch add_gripper:=true <run_demo:=true>
```

nkise, 2020-05-25:
@penglongxiang I've checked the last commit and it seems working now. Can I control gripper now in the gazebo or you didn't try that feature?

penglongxiang, 2020-05-26:
For the time being, we have not tried to control the Gripper in simulation yet, you can try to write a controller node to give command to drive_joint of the gripper. We will start the gripper simulation development soon.