xArm-Developer/xarm_ros #20: Unable to identify any set of controllers that can actuate the specified joints: [ joint1 joint2 joint3 joint4 joint5 joint6 joint7 ]

jungladicitta, 2020-05-26:
Hello!
when I spawn xarm7 in gazebo:
`roslaunch xarm_gazebo xarm7_beside_table.launch `
I see errors:
```
[ERROR] [1590488171.835015000, 0.394000000]: No p gain specified for pid.  Namespace: /xarm/gazebo_ros_control/pid_gains/joint1
[ERROR] [1590488171.836314999, 0.394000000]: No p gain specified for pid.  Namespace: /xarm/gazebo_ros_control/pid_gains/joint2
[ERROR] [1590488171.837444478, 0.394000000]: No p gain specified for pid.  Namespace: /xarm/gazebo_ros_control/pid_gains/joint3
[ERROR] [1590488171.838585170, 0.394000000]: No p gain specified for pid.  Namespace: /xarm/gazebo_ros_control/pid_gains/joint4
[ERROR] [1590488171.839765461, 0.394000000]: No p gain specified for pid.  Namespace: /xarm/gazebo_ros_control/pid_gains/joint5
[ERROR] [1590488171.840988391, 0.394000000]: No p gain specified for pid.  Namespace: /xarm/gazebo_ros_control/pid_gains/joint6
[ERROR] [1590488171.841756028, 0.394000000]: No p gain specified for pid.  Namespace: /xarm/gazebo_ros_control/pid_gains/joint7
```
and then when I launch:
`roslaunch xarm7_moveit_config xarm7_moveit_gazebo.launch`
and click on Execute in rviz I see errors:
```
[ERROR] [1590489570.573187541, 280.419000000]: Unable to identify any set of controllers that can actuate the specified joints: [ joint1 joint2 joint3 joint4 joint5 joint6 joint7 ]
[ERROR] [1590489570.573237097, 280.419000000]: Known controllers and their joints:
```
I tried to set pid values in `~/catkin_ws/src/xarm_ros/xarm7_moveit_config/config/xarm7_params.yaml`. Errors `No p gain specified for pid` disappeared, but model in gazebo starts falling down and then teleports into the floor under and behind the table.

how can I solve this problem to execute plan in rviz? which pid values should I set?

penglongxiang, 2020-05-26:
@jungladicitta The proper pid gains are not easy to determine, you can just leave the "no pid gains" errors there, and nothing is going to be affected. 

As for "Unable to identify any set of controllers", have you followed chapter 3 of ReadMe to install **ros-controllers** and **ros-control**? You can check by:  
```bash
dpkg -l | grep "ros-control"
```

jungladicitta, 2020-05-26:
@penglongxiang thanks, I forgot to install moveit.

closed 2020-05-26 by jungladicitta

jungladicitta, 2020-05-26:
@penglongxiang 
sorry for offtopic questions,
1) is gripper available for xarm7?
2) is rgbd camera supported in gazebo on xarm or I should attach it manually?

penglongxiang, 2020-05-27:
@jungladicitta currently the gripper is available only as a 3D model, the controller of gripper is not developed yet. For simulation, you can refer to the xarm6 example and attach the gripper to xarm7 by yourself. 

There is no example or model for cameras in our packages, you need to implement by yourself as well.