xArm-Developer/xarm_ros #21: xarm last updates and how-to at the end of May 2020

jungladicitta, 2020-05-27:
1. gripper in gazebo.
I launch 
`roslaunch xarm_gazebo xarm6_beside_table.launch add_gripper:=true`
and
`roslaunch xarm6_gripper_moveit_config xarm6_gripper_moveit_gazebo.launch`
and then in rviz when I choose planning group to xarm_gripper and click on Plan&Execute random valid goal I see error: `Unable to identify any set of controllers that can actuate the specified joints: [ drive_joint ]`
```
[ INFO] [1590564017.938652113, 33.026000000]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1590564017.939214614, 33.027000000]: Planner configuration 'xarm_gripper' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1590564017.939659265, 33.027000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1590564017.939719885, 33.027000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1590564017.939799389, 33.027000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1590564017.939940968, 33.027000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1590564017.953521355, 33.041000000]: RRTConnect: Created 7 states (2 start + 5 goal)
[ INFO] [1590564017.953644197, 33.041000000]: RRTConnect: Created 7 states (2 start + 5 goal)
[ INFO] [1590564017.955738016, 33.041000000]: RRTConnect: Created 7 states (2 start + 5 goal)
[ INFO] [1590564017.955850722, 33.041000000]: RRTConnect: Created 8 states (2 start + 6 goal)
[ INFO] [1590564017.956544702, 33.041000000]: ParallelPlan::solve(): Solution found by one or more threads in 0.017047 seconds
[ INFO] [1590564017.956943353, 33.045000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1590564017.957010164, 33.045000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1590564017.958066013, 33.045000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1590564017.958191260, 33.046000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1590564017.960608571, 33.048000000]: RRTConnect: Created 7 states (2 start + 5 goal)
[ INFO] [1590564017.961609250, 33.049000000]: RRTConnect: Created 7 states (2 start + 5 goal)
[ INFO] [1590564017.962110127, 33.050000000]: RRTConnect: Created 7 states (2 start + 5 goal)
[ INFO] [1590564017.964275938, 33.052000000]: RRTConnect: Created 8 states (2 start + 6 goal)
[ INFO] [1590564017.965227264, 33.053000000]: ParallelPlan::solve(): Solution found by one or more threads in 0.008469 seconds
[ INFO] [1590564017.965442032, 33.053000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1590564017.965487039, 33.053000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1590564017.968904286, 33.057000000]: RRTConnect: Created 7 states (2 start + 5 goal)
[ INFO] [1590564017.968997145, 33.057000000]: RRTConnect: Created 7 states (2 start + 5 goal)
[ INFO] [1590564017.970238554, 33.057000000]: ParallelPlan::solve(): Solution found by one or more threads in 0.004848 seconds
[ INFO] [1590564018.000741571, 33.085000000]: SimpleSetup: Path simplification took 0.030395 seconds and changed from 6 to 2 states
[ INFO] [1590564026.934026733, 41.966000000]: Execution request received
[ INFO] [1590564026.934113786, 41.966000000]: Returned 1 controllers in list
[ERROR] [1590564026.934164558, 41.966000000]: Unable to identify any set of controllers that can actuate the specified joints: [ drive_joint ]
[ERROR] [1590564026.934221128, 41.966000000]: Known controllers and their joints:
controller '' controls joints:
  joint1
  joint2
  joint3
  joint4
  joint5
  joint6

[ INFO] [1590564026.943665812, 41.975000000]: ABORTED: Solution found but controller failed during execution
```
what does this error mean? Is gripper manipulation available in gazebo?

2. gripper teleoperation. How can I do gripper teleoperation in gazebo?
I tried to launch
`roslaunch xarm_gripper gazebo.launch` and
`roslaunch xarm_gripper gripper_rviz_display.launch `
but it didnt work.

3. gripper and xarm7. is gripper supported with xarm7? 

4. xarm and camera in simulation.
How can I attach camera to gripper in gazebo simulation? Is there any launch file provided or I can do it manually?

penglongxiang, 2020-05-27:
Please see the reply in [here](https://github.com/xArm-Developer/xarm_ros/issues/20#issuecomment-634482450)

penglongxiang, 2020-06-19:
hi @jungladicitta xArm gripper simulation is better supported now, please check the latest commit.   
 
Planning group "xarm_gripper" can be controlled in Moveit-Gazebo simulation. You can define more group_states ("open"/"close" already provided) in [xarm6_with_gripper.srdf](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm6_gripper_moveit_config/config/xarm6_with_gripper.srdf) for the gripper.   

 xArm5 and xArm7 with gripper moveit packages are also developed.  

For real gripper control, actions and services are supported. but can not be controlled through moveit planner. Please see the [ReadMe](https://github.com/xArm-Developer/xarm_ros#gripper-control) for more info.