xArm-Developer/xarm_ros #25: Position feedback rate

xqms, 2020-06-09:
Hi there,

is there some way to increase the frequency of the position feedback? We are using the xArm6 in a low-latency application. While we can provide commands with 100 Hz, the position feedback is only 5 Hz. It would be great if this could be improved :)

penglongxiang, 2020-06-10:
Hi @xqms, For now, only 5Hz of feedback is supported in provided ros interfaces. If you have [xArm Developer Manual](http://download.ufactory.cc/xarm/en/xArm%20Developer%20Manual.pdf?v=1591725074801) at hand, refer to chapter 3.1.3.【Automatic Reporting Format】(page 37), you can try to analyze the reported data stream on port 30003, which updates at 100Hz rate and contains joint position feedback.

xqms, 2020-12-09:
Hi @penglongxiang, is it possible that the stream on port 30003 reports *commanded* joint positions, not measured joint positions? I'm seeing some strange timing issues which could be explained that way...

penglongxiang, 2020-12-10:
hi @xqms , the report is currently not configurable, may I ask what is the exact timing issue?

xqms, 2020-12-10:
Well, I haven't done precise tests yet. But I was trying to measure the latency between the report joint positions and the robot's true position (by placing a camera on the end of the arm and synchronizing the images to the movement). It seems that the reported position arrives at the goal around 70ms before the arm itself does... But I have to make more precise measurements (and compare with the commanded position) later today.

penglongxiang, 2020-12-10:
Hi @xqms I just did some check and the port 30003 actually reports the COMMAND positions while robot is under position control. So that may explain why the report position is slightly ahead of actual position.

xqms, 2020-12-10:
Thank you for checking so quickly :+1:

That explains what I'm seeing and now that I know it I can work with it :) Still, it would be very nice to get the actual measured position somehow. Can you think of a way to do that?

penglongxiang, 2020-12-11:
We will take this advice into consideration and add this option in future versions if there is a way.

xqms, 2020-12-12:
Ok, thank you again for the very fast response time :)

closed 2020-12-12 by xqms

lenzc, 2023-06-29:
Hello,
is there any update regarding receiving MEASURED joint states?
Best
Chris

penglongxiang, 2023-06-30:
Hi @lenzc, there will be a full-state reporting port added in the next firmware release version v2.1.0 (including both command and actual position/velocity data, updated at 200Hz).  Hopefully it will be released in a week. 

Meanwhile you can use our python API [get_servo_angle()](https://github.com/xArm-Developer/xArm-Python-SDK/blob/0fd107977ee9e66b6841ea9108583398a01f227b/xarm/wrapper/xarm_api.py#L870) and specify `is_real=True` to actively get the measured joint positions. Currently there is no direct feedback of MEASURED joint states in ROS.

lenzc, 2023-06-30:
Hi @penglongxiang, thanks for the fast and helpful reply. Looking forward to the firmware release.

lenzc, 2023-08-15:
Hi @penglongxiang, do you have any update regarding the firmware release v.2.1.0?

penglongxiang, 2023-08-21:
Hi @lenzc, the firmware v2.1.0 is already available with the full-state report function at port 30000. Please check the content information [here](https://github.com/xArm-Developer/ufactory_docs/blob/main/firmware/report/ReadMe_en.md), and [sample python code](https://github.com/xArm-Developer/ufactory_docs/blob/main/firmware/report/get_report_data_with_port30000.py) for getting certain field data with port 30000. Please update the controller firmware and test this function on your side.

Currently the report rate is 200Hz, and we are still estimating if the rate can be further increased to 250Hz max, we will update the instruction if there is any change in the future.

lenzc, 2023-08-22:
Hi @penglongxiang, thanks for the quick reply. I updated the arm and was able to read measured joint positions using the Python API. However, I would like to use the full ROS stack with the measured joint position. Is there any plan to add port 30000 in [xArm-CPLUS-SDK](https://github.com/xArm-Developer/xArm-CPLUS-SDK/blob/7d4e1ba7782a016dcf7d00b56f29a88f97cbea46/include/xarm/core/report_data.h) or is there another way to get the measured joint positions in ROS?