xArm-Developer/xarm_ros #27: xarm_gripper build problem

ldemk, 2020-07-02:
Hello 
The package xarm_gripper fails when `catkin build`. I fixed the error. How can I commit it here and create pull request?

penglongxiang, 2020-07-02:
Hi @ldemk, thanks for finding the issue. We use "catkin_make" usually. 

To make a pull request, I believe you need to **fork** this repository first, commit and push to your forked online repo, then you may see something like "compare and pull request" button to this official repo.

closed 2020-07-05 by penglongxiang

TZVIh, 2020-07-08:
Hi @ldemk 
Can you please tell me how did you fix this issue?
Thanks!

ldemk, 2020-07-09:
Hello @TZVIh 
I added the xarm_msgs package to package.xml and CMakeList.txt files and it worked. You can see by yourself my pull request #28. You just have to pull and it will work on your side also.
By the way, I noticed other issues of yours, so I'd like to point out that I am using Kinetic.

PR #28 (2020-07-05): added xarm_msgs
This is the fix of the problem with `catkin build` of the package `xarm_gripper`

added `xarm_msgs` to `package.xml` and `CMakeLists.txt`

This closes #27