xArm-Developer/xarm_ros #34: Sudden jump when switching to mode 1.

RexDrac, 2020-07-31:
Hi,

When I try to switch from mode 0 to mode 1 or from mode 2 to mode 1 to control the robot with moveit, the robot performs a very jerky jump at the moment of the switch.

The jump only occurs when I attempt to switch to mode 1. There are no problems when switching from mode 1 to mode 0 or mode 1 to mode 2.

Are there any ways to solve this problem?

Many thanks

penglongxiang, 2020-08-03:
Hi, sorry about the issue and bad experience, may I get more information about how this sudden jump was triggered? Is there any background process (like moveit node) still running when you switch from mode 2 to mode 1?

Normally when switching to mode 2, it is recommended to stop any controlling process of xArm, For example, if there is moveit controller running at the background,  it will continuously send position command in mode1 at 100Hz even in stationary state. When actual robot position is changed in mode 2 but moveit controller wouldn't know that and will still send previous position to xArm. Then it will be likely to jump when mode is switched from 2 back to 1.  

This is just a possibility from my guess, could you provide the detailed operations (and order) you performed? We will help identify the true cause and provide solutions more quickly. Thank you very much.

RexDrac, 2020-08-04:
Yes, there is a moveit process running on the background.

I am able to bypass the problem by disabling the motor before switching to mode 1, and then enabling back the motor after the switch.

Are there any methods to inform the moveit controller about the change in position? I would like to be able to freely switch between different modes, therefore I would like to avoid terminating the moveit process whenever I switch.

Many thanks.

penglongxiang, 2020-08-06:
Hi @RexDrac ,

I am not totally sure about your situation, but you can try updating and compiling your xarm_ros to the latest version, which will reset the controller when a mode switch or error occurs. Then after switching back to mode 1, be sure to update start state to current state (you can do it in moveit rviz panel or use [setStartStateToCurrentState ()](http://docs.ros.org/kinetic/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1MoveGroupInterface.html#a34e857711a8dca3e98635f4925abaf46) in MoveGroupInterface API) before any new motion plans or executions.

Please check it with caution and a small position change in Mode 2. Hope this can help.