xArm-Developer/xarm_ros #35: Gripper control through MoveIT with real xarm7 not working

mmatsi, 2020-08-20:
Hello,
I've faced with a problem when trying to control real Xarm7 gripper through MoveIT.
First of all, a real gripper position is not visualized with rviz, second - commands from moveit are not executed on real hand: planning is successful, but the execution fails with:
```bash
[ERROR] [1597913128.002681382]: Unable to identify any set of controllers that can actuate the specified joints: [ drive_joint ]
[ERROR] [1597913128.002724845]: Known controllers and their joints:
controller 'arm_controller' controls joints:
  joint1
  joint2
  joint3
  joint4
  joint5
  joint6
  joint7
```

Commands executed to start moveit:
```bash
roslaunch xarm7_gripper_moveit_config realMove_exec.launch robot_ip:=<your controller box LAN IP address>
```
Info about the system:
Laptop:
  Ubuntu 16.04
  ROS Kinetic
Control box:
  Firmware: 1.5.2
  Xarm studio: 1.5.2

penglongxiang, 2020-08-20:
@mmatsi We did try to develop Moveit support for the Gripper. However, we found real-time communication to the gripper is too costly and may influence the performance of xArm in Moveit control. So gripper will not be able to be directly controlled by Moveit in this generation. Instead you can use ROS service to control the real gripper, see "[5.7.7 Gripper Control](https://github.com/xArm-Developer/xarm_ros#gripper-control)" in ReadMe. Sorry about the inconvenience.

penglongxiang, 2020-08-20:
The gripper model loaded in moveit will play a role in self-collision prevention, when controlling real arm with a gripper installed .

vimior, 2023-03-10:
@BhooshanDeshpande 
Currently both xarm_ros and xarm_ros2 can control xarm_gripper through moveit.