xArm-Developer/xarm_ros #36: Controller handle arm_controller reports status PREEMPTED

yfedi, 2020-08-25:
Hello,

I'm working with arm and time to time I receive next messages, when arm is moving. After that I need to restart node
```
[ INFO] [1598371134.336988967]: SimpleSetup: Path simplification took 0.007873 seconds and changed rom 4 to 2 states
[ INFO] [1598371135.875451622]: Execution request received
[ WARN] [1598371140.829753677]: Controller handle arm_controller reports status PREEMPTED
[ INFO] [1598371140.829787962]: Completed trajectory execution with status PREEMPTED ...  
[ INFO] [1598371140.829833516]: Execution completed: PREEMPTED  
```
What this issue means and how it could be resolved?

Commands executed to start moveit:

`roslaunch xarm7_gripper_moveit_config realMove_exec.launch robot_ip:=<your controller box LAN IP address>`

Info about the system:
Laptop:
Ubuntu 16.04
ROS Kinetic
Control box:
Firmware: 1.5.2
Xarm studio: 1.5.2

penglongxiang, 2020-08-26:
Hi, did you see any error report in topic /xarm/xarm_states ?

yfedi, 2020-08-26:
Hello, 
Thank you for the fast response, I will check topic /xarm/xarm_states and provide updates.

yfedi, 2020-08-27:
In topic /xarm/xarm_states next message
`
state: 4
mode: 0
cmdnum: 0
mt_brake: 127
mt_able: 127
err: 23
warn: 0
angle: [1.222000002861023, 2.0969998836517334, -2.4839999675750732, 1.7790000438690186, 2.441999912261963, 2.1610000133514404, -2.678999900817871]
pose: [464.76220703125, 448.8066101074219, 204.22593688964844, 3.1399950981140137, 0.011669198982417583, 3.0873985290527344]
offset: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
`
Where I can find information about errors?

penglongxiang, 2020-08-28:
Error 23 means there is at least one joint position exceeding the normal range, in your case, it is joint 2. Please refer to our [user manual](https://cdn.shopify.com/s/files/1/0012/6979/2886/files/xArm_User_Manual.pdf?v=1597982519) for error meaning (section 1.2.1 in page 153) and joint operating range (section 2.4 in page 161). 

If error occurs during **moveit application**, after handling the error. You have to restart the application or follow [these steps](https://github.com/xArm-Developer/xarm_ros#62-proper-way-to-change-modes) to switch back to **mode 1** to continue. You may need to update the starting position as "current position" to ensure next successful planning.