xArm-Developer/xarm_ros #38: Questions on dual arm control

RexDrac, 2020-09-11:
Hi,

I have a few questions on dual arm control.

1. I would like to ask if it is possible to control dual arms in mode 0 through rosservice commands? Or is the dual arm control only possible in mode 1 using moveit?

2. Is it possible to let the dual arm work and switch to different modes seperately from each other? For example, having the left arm working in mode 2, while the right arm is working in mode 1.

Thanks.

penglongxiang, 2020-09-14:
Hi, if you are not planning to use moveit, you can manage to control two xArms in mode 0 separately by launching xarm_bringup/xarm6_server.launch in different namespaces, refer to [this file](https://github.com/xArm-Developer/xarm_ros/blob/master/examples/multi_xarm5/launch/two_real_xarm6.launch).  The reason to use Moveit in  [this example](https://github.com/xArm-Developer/xarm_ros/tree/master/examples#4-dual-xarm6-controlled-with-one-movegroup-node) is that collision can be avoided in dual arm planning in Moveit framework.

gibernas, 2021-03-09:
Hi @penglongxiang , I am setting up a system so that it works as intented in the first part of your reply, as multiple robot should be launched simultaneously but in different working areas.

I tried with simply using a namespace group and passing `ext_ns` as done in the first file, but then some other launch files are included using a different namespace, coming from `xarm_hw_namespace`, resulting in a nested namespace with `ext_ns/xarm_hw_namespace`. 

As a result i get for example the `joint_state_publisher` to be published at `/ext_ns/xarm_hw_namespace/joint_states` but subscribed at `/ext_ns/joint_states`. However if I try to remove the innermost namespace, the driver node dies at launch.

What is the difference between the two namespace arguments? I don't get what I am doing wrong...