xArm-Developer/xarm_ros #39: Xarm7: Joint 6 Issue

mmatsi, 2020-09-22:
Hello,
During usage of Xarm7 I am constantly heading with the following error:
[image]
This error comes up during usage of MoveIT, when controlling hand via Xarm Studio (web interface) and even when using joint teach mode. What makes hand usage almost impossible

zfls, 2020-09-22:
Hello,
You can solve the issue through the following steps:
1) power off the xArm(press the emergency stop button)
2) power off the control box 
3) power on the control box
4) power on the xArm(rotate the emergency stop button)

If the problem still exists, please send the SN of the xArm base to us,thanks.

Best,

mmatsi, 2020-09-22:
Thx for the quick reply
Unfortunatelly, it doesnt solve the problem. Could you please give me an email, where I can send the serial number. The serial number is located on the sticker control box?

zfls, 2020-09-22:
Hello,
You can send the SN of the xArm to my email: shiyanfei@ufactory.cc
The SN of the xArm is shown in the picture below:
[image]

closed 2020-10-22 by penglongxiang