xArm-Developer/xarm_ros #40: noetic support

AndreV84, 2020-09-29:
is noetic ROS supported?

penglongxiang, 2020-09-30:
We have not yet tested on ROS noetic, you can give it a try, but it is not guaranteed to be fully functional.

swiz23, 2020-12-21:
I was able to successfully compile all the ROS packages in ROS Noetic. I didn't test all of them, but I confirmed that the MoveIt package (for the xarm6) works correctly.

To make it fully compatible, 

- the 'use_gui' parameter for the joint_state_publisher should be removed and instead, the joint_state_publisher or the joint_state_publisher_gui should be launched where appropriate.

- The '--inorder' tag when loading from xacro should be removed.

- The kinematics_attempts variable in the moveit config file should be removed

JuliusSustarevas, 2021-02-08:
Are you planning on PRing this? @penglongxiang  are PRs accepted in this repo?

penglongxiang, 2021-02-12:
@JuliusSustarevas I've not received PRs for above mentioned points. So far as I have tested with Noetic, this repo can still work directly, although with some warnings of those deprecated parameters.

swiz23, 2021-02-12:
Hi @JuliusSustarevas,

I don't have plans to make a PR currently. That said, I created my own version of these ROS packages that has full Kinetic, Melodic, and Noetic support which can be found at https://github.com/Interbotix/interbotix_ros_manipulators/tree/main/interbotix_ros_uxarms.