xArm-Developer/xarm_ros #43: Cannot control via MoveIt!

TMats, 2020-12-01:
Thanks for developing the packages!

In the current master branch at https://github.com/xArm-Developer/xarm_ros/commit/055b1bab1ac7c7966884c2797db09f42730e2ae9, real robots cannot be controlled through MoveIt (execution fails).

The command I launched MoveIt is 
```
$ roslaunch xarm7_gripper_moveit_config realMove_exec.launch robot_ip:=<your controller box LAN IP address>
```
as described in https://github.com/xArm-Developer/xarm_ros#to-run-moveit-motion-planner-to-control-the-real-xarm-with-xarm-gripper-attached.

And the error occurs as below,

```
[ INFO] [1606814705.233160442]: ABORTED: Solution found but controller failed during execution
[ INFO] [1606814806.654758866]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1606814806.656490789]: Planner configuration 'xarm7[RRTConnect]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1606814806.658114161]: xarm7[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1606814806.658432332]: xarm7[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1606814806.658749222]: xarm7[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1606814806.659127183]: xarm7[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1606814806.674514077]: xarm7[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1606814806.674824208]: xarm7[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1606814806.675870649]: xarm7[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1606814806.676187360]: xarm7[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1606814806.676898691]: ParallelPlan::solve(): Solution found by one or more threads in 0.019365 seconds
[ INFO] [1606814806.678005553]: xarm7[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1606814806.678423674]: xarm7[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1606814806.678815814]: xarm7[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1606814806.679104555]: xarm7[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1606814806.684202083]: xarm7[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1606814806.684395253]: xarm7[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1606814806.685232214]: xarm7[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1606814806.685583275]: xarm7[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1606814806.686071616]: ParallelPlan::solve(): Solution found by one or more threads in 0.008401 seconds
[ INFO] [1606814806.687052387]: xarm7[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1606814806.687277807]: xarm7[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1606814806.693070867]: xarm7[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1606814806.693313877]: xarm7[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1606814806.694019768]: ParallelPlan::solve(): Solution found by one or more threads in 0.007306 seconds
[ INFO] [1606814806.712420117]: SimpleSetup: Path simplification took 0.018009 seconds and changed from 3 to 2 states
[ INFO] [1606814807.706837994]: Execution request received
[ INFO] [1606814807.707018114]: Returned 0 controllers in list
[ERROR] [1606814807.707113044]: Unable to identify any set of controllers that can actuate the specified joints: [ joint1 joint2 joint3 joint4 joint5 joint6 joint7 ]
[ERROR] [1606814807.707210575]: Known controllers and their joints:
```

However, I confirmed this command works in https://github.com/xArm-Developer/xarm_ros/commit/472f475d2d4a3b2cd6fd34e1ce7a024c9c416045, by checking out to the commit.

I doubt some changes to moveit control have been added between these commits, but I cannot find them.
Do you have any idea on the solution or reasons?

Thanks again for the active development!

penglongxiang, 2020-12-02:
Hi @TMats , thank you so much for the bug report. We have fixed this issue with namespace correction, please check out the latest commit and have a try.

TMats, 2020-12-02:
@penglongxiang It worked! Thank you very much for the rapid response!

closed 2020-12-02 by TMats