xArm-Developer/xarm_ros #45: Safety Tips

swiz23, 2020-12-08:
Hi,

I was just wondering what would happen if you called one of the ROS services (move_line, move_joint, mve_servo_cart, or move_servoj) to move the robot with either an invalid joint goal (one or more of the desired joint positions are above or below the URDF limits), or an invalid pose goal (either due to joint limits being violated or because the goal is physically impossible). I don't want to try this out on the physical robot in case I break it, so I'm asking it here. Does the internal robot controller take this into account no matter if you are in Mode 0 or Mode 1, or is this something the user must check before commanding the goals?

Thanks

penglongxiang, 2020-12-09:
Hi @swiz23, for invalid/unreachable commands, the arm will report an error before hitting the singularity or joint limit position. In case of damaging the robot arm, please make sure the firmware version is 1.6.0 or later, so that the robot can monitor its configuration and report an error (Code 22) when it is about to hit itself, and by setting the proper collision sensitivity(>0), mounting type as well as payload parameters, the robot can perform a protective stop even a collision with the environment did happen.

swiz23, 2020-12-09:
Awesome. Thanks!

closed 2020-12-09 by swiz23