xArm-Developer/xarm_ros #46: Calling the set_tcp_offset ROS service

swiz23, 2020-12-10:
For whatever reason, after calling this service, the robot's state (as shown in the xarm_states topic) changes to '5'. The error code is 0, but if I ever call this service, I have to reset the state to 0 to work with the robot. From the RobotMsg message description, it doesn't even show '5' as an option (only 1 - 4). What does this mean, and why does it happen?

penglongxiang, 2020-12-10:
Hi @swiz23 , sorry I forgot to update the message description, state 5 is newly added for Configuration Change state, which happens after some critical configuration change (switch mode, offset/payload change, etc), which will clear the buffered command and require a system reset (set_state=0) before next motion.

swiz23, 2020-12-10:
Got it. Thanks!

closed 2020-12-10 by swiz23