xArm-Developer/xarm_ros #47: Bug When Switching Modes using ROS Service

swiz23, 2020-12-14:
There is no issue when switching from Mode 0 to Mode 1 or from Mode 0 to Mode 2 or from either of those modes to Mode 0.

However, when switching from Mode 1 to Mode 2 or from Mode 2 to Mode 1, the Mode automatically goes back to Mode 0.

I noticed this issue only occurs if the state in the xarm_states topic is set to '2'. But, if the state in the xarm_states topic is set to '5', then there is no issue.

To clarify...
In the image below, the robot is in Mode 2 and has a state of 2.
[image]

Then I type `rosservice call /xarm/set_mode "data: 1"` in a terminal. The result is 'ret: 0' and the message is 'message: "servo mode, ret = 0".

However, the robot in actuality goes to Mode 0 as shown in the topic below...
[image]

Now - if the the robot is in Mode 2 and has a state of 5.
[image]

Then I type the rosservice call above to change to mode 1, it works correctly.
[image]

Why is this happening?

penglongxiang, 2020-12-16:
Hi @swiz23 , thank you very much for the detailed bug report. Mode 0 is the main and default control mode, sometimes it will switch to this mode automatically (for example, when error occurs) in order to be safe. Clearly there is some logic issue for the situation you mentioned, we will try to correct this in the next firmware update.

swiz23, 2020-12-18:
Good to hear

SMRazaRizvi96, 2024-01-26:
Any update on this?

I am having a similar issue.

My Arm randomly gets into state:1 and mode:1, however I want it to start in state:2 mode:1.

Then, even if I trigger the state change request to 2, nothing happens.