xArm-Developer/xarm_ros #48: xArm for ROS2

EgOrlukha, 2020-12-16:
Dear xArm developers,

thanks for your awesome product, it is a great solution for needs of our team.
Could you tell, is the current xarm_ros package compatible with ROS2? If no, do you plan transferring your progress to ROS2?

Thanks in advance!

penglongxiang, 2020-12-17:
Hi @EgOrlukha ,

Glad you like our product and thank you for your support. Currently xarm_ros is not developed for ROS2 yet, we will do an evaluation and if there is a plan on supporting ROS2 next year, we will put up a notice in this repository.

lorepieri8, 2021-02-19:
@penglongxiang Any updates on xArm on ROS2?

penglongxiang, 2021-02-20:
@lorepieri8 We have development plan for ROS2 this year. However the release time is uncertain for now.

penglongxiang, 2021-06-17:
@EgOrlukha @lorepieri8 Now the first version of  xArm for ROS2 is released. Please feel free to check and try out. Repository is at: (https://github.com/xArm-Developer/xarm_ros2). Please note that not all functions in xarm_ros are implemented as official ROS2 and related Moveit2 & Gazebo support are still under development.

closed 2021-06-23 by penglongxiang