xArm-Developer/xarm_ros #49: TCP Offset & MoveIt

micahpearlman, 2020-12-16:
When setting the TCP offset via `/xarm/set_tcp_offset` MoveIt does not appear to recognize this when planning.  Is this by design?

penglongxiang, 2020-12-17:
Hi @micahpearlman , the provided ros services are only effective for our xArm controller planning algorithm, since Moveit is a third party motion planning software, you probably need to find a way to make Moveit aware of the additional offset.

penglongxiang, 2021-01-06:
Hi @micahpearlman we have made some modifications on xarm urdf and moveit configuration srdf file, when running 'roslaunch  xarm6_gripper_moveit_config demo.launch', the marker in rviz now appears at gripper end point rather than the robot tool flange. I think this is the proper way, to add additional link to model description file that Moveit loads.