xArm-Developer/xarm_ros #50: Bug in ROS xarm_driver_node at shutdown

swiz23, 2020-12-16:
Hi,

When running `roslaunch xarm_bringup xarm6_server.launch robot_ip:=<robot_ip>` on the latest code, everything starts up and runs fine. However, when `Ctrl-C`ing the launch (which is only running the xarm_driver_node), I get the following error message - something to do with throwing an instance of 'std::system_error'. I think it's related to threading, but not sure how.

[image]

penglongxiang, 2020-12-17:
Thank you again for the report. 'xarm_driver' is running in multi-threading, I think we need to catch the Ctrl-C signal and handle this exception clean and properly. This issue would not actually affect the normal execution of the program, we will handle this in the next updates.

swiz23, 2020-12-18:
Sounds good

penglongxiang, 2020-12-22:
Hi, this error should be resolved by the latest commit. Please have a check.

closed 2020-12-22 by penglongxiang