xArm-Developer/xarm_ros #51: Compiling error from the source code.

AnsunWu, 2020-12-17:
Hello.

I followed the x-arm 'read. me' file and installed all required dependencies.
But when I compiled the workspace, two errors show up and the compile was failed.
The error message is shown below, Would you please help me to check it? and how to solve this issue?
My ROS version is "ROS 16.04 kInetic", Thanks!!!!

`[ 91%] Built target test_tool_modbus
[ 91%] Built target sample_motion
In file included from /usr/include/eigen3/Eigen/Core:297:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/kinetic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_state/robot_state.h:41,
                 from /opt/ros/kinetic/include/moveit/move_group_interface/move_group_interface.h:43,
                 from /home/anson/xarm_github/src/xarm_ros/xarm_planner/src/xarm_gripper_planner.cpp:7:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/opt/ros/kinetic/include/moveit_visual_tools/moveit_visual_tools.h:227:45:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:330:5: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
In file included from /usr/include/eigen3/Eigen/Core:297:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/kinetic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_state/robot_state.h:41,
                 from /opt/ros/kinetic/include/moveit/move_group_interface/move_group_interface.h:43,
                 from /home/anson/xarm_github/src/xarm_ros/xarm_planner/src/xarm_simple_planner.cpp:7:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/opt/ros/kinetic/include/moveit_visual_tools/moveit_visual_tools.h:227:45:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:330:5: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
[ 91%] Linking CXX executable /home/anson/xarm_github/devel/lib/xarm_api/xarm_driver_node
[ 91%] Built target xarm_driver_node
xarm_ros/xarm_planner/CMakeFiles/xarm_gripper_planner.dir/build.make:62: recipe for target 'xarm_ros/xarm_planner/CMakeFiles/xarm_gripper_planner.dir/src/xarm_gripper_planner.cpp.o' failed
make[2]: *** [xarm_ros/xarm_planner/CMakeFiles/xarm_gripper_planner.dir/src/xarm_gripper_planner.cpp.o] Error 1
CMakeFiles/Makefile2:5552: recipe for target 'xarm_ros/xarm_planner/CMakeFiles/xarm_gripper_planner.dir/all' failed
make[1]: *** [xarm_ros/xarm_planner/CMakeFiles/xarm_gripper_planner.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
xarm_ros/xarm_planner/CMakeFiles/xarm_simple_planner.dir/build.make:62: recipe for target 'xarm_ros/xarm_planner/CMakeFiles/xarm_simple_planner.dir/src/xarm_simple_planner.cpp.o' failed
make[2]: *** [xarm_ros/xarm_planner/CMakeFiles/xarm_simple_planner.dir/src/xarm_simple_planner.cpp.o] Error 1
CMakeFiles/Makefile2:6097: recipe for target 'xarm_ros/xarm_planner/CMakeFiles/xarm_simple_planner.dir/all' failed
make[1]: *** [xarm_ros/xarm_planner/CMakeFiles/xarm_simple_planner.dir/all] Error 2
`

penglongxiang, 2020-12-17:
Hi @AnsunWu , it seems to be an incompatible issue with your local Eigen3 and moveit versions. Did you install moveit from source? I checked the web and found [this similar issue](https://github.com/ros-planning/moveit_tutorials/issues/262), maybe you can take a look and see if this error would disappear after checking out the correct version of moveit source code.

AnsunWu, 2021-01-03:
Hello, Thanks for your reply, I compiled the Moveit! package from the source and the problem is solved.

closed 2021-01-03 by AnsunWu

penglongxiang, 2021-01-27:
Just found out this issue can be solved by changing `Eigen::Affine3d` to `Eigen::Isometry3d` according to [this reply](https://github.com/ros-planning/moveit_tutorials/issues/262#issuecomment-444896572). The latest commit have handled this problem. No need to compile Moveit from source in the future.