xArm-Developer/xarm_ros #53: Mimic Joint Plugin Update

swiz23, 2020-12-18:
Hi,

I noticed that the mimic joint plugin used in this repo has actually been deprecated by the maintainer a few days ago. It is now located at https://github.com/roboticsgroup/roboticsgroup_upatras_gazebo_plugins. The plugin's name has also changed to include 'upatras' in the filename. Not sure what the difference between the versions are, but I did see that the plugin is now available as a debian package from apt (the previous version wasn't).

That said, the ros-kinetic-roboticsgroup-upatras-gazebo-plugins debian package has dependencies on gazebo7, so you will still have to build the new plugin from source on ROS Kinetic to make it work with Gazebo 9. I've already tried this and confirmed that it works.

penglongxiang, 2020-12-22:
Thank you very much for this notice. I have tested and mentioned this issue in [ReadMe](https://github.com/xArm-Developer/xarm_ros#34-install-mimic_joint_plugin-for-xarm-gripper-simulation-in-gazebo). The former version can still work if previously downloaded and compiled.

swiz23, 2020-12-24:
Awesome

closed 2020-12-24 by swiz23