xArm-Developer/xarm_ros #55: Open up Joint State Publishing Frequency as a ROS Parameter

swiz23, 2020-12-21:
...instead of having it hardcoded in the xarm_driver_node to 10 as shown at https://github.com/xArm-Developer/xarm_ros/blob/34e42794598c3ac2eefce37cf82767dbc9c85718/xarm_api/src/xarm_driver_node.cpp#L138

That way, it would be easier for me to create a ROS node that does record and playback.

penglongxiang, 2020-12-22:
Hi, currently the actual full-state report from xArm controller changes only in 5Hz, we did not make it a ros parameter because there is not much margin for adjustment. You may modify your local code for your convenience. Or if you want a higher feedback rate, check our [developer manual](https://cdn.shopify.com/s/files/1/0012/6979/2886/files/xArm_Developer_Manual-1.6.0.pdf?v=1607666925), and search for "REPORT_TCP_DEVELOP", that is to connect the controller IP at port 30003, and decode the data for critical joint states reporting at 100Hz.

swiz23, 2020-12-24:
So what's running at 10hz in the code there?

penglongxiang, 2020-12-25:
10Hz is the frequency to check the update of xArm full-state report.

JuliusSustarevas, 2021-02-12:
Sorry, I'm just evaluating if the arm is suitable for me. Can you tell me what information is inside this report? Does this mean xarm joint-states are being updated only at 5Hz? that seems to be incredibly slow, is there a plan to address this in the future? 
Presumably the controllers are able to control it much faster, the joint states here are just how fast it publishes to outside the control box? 
 I plan todo disturbance rejection using the arm to stabilise the end-effector.  Will the arm be able to be controlled effectively at 50-100Hz?

penglongxiang, 2021-02-12:
@JuliusSustarevas  For more information about the 5Hz report, please check the developer manual link in my previous answer at page 66 "REPORT_TCP_NORMAL". Inside our controller, the feedback and control rate are 250Hz, however currently in ROS  joint_state interface we use a 5Hz automatic state report. User can still get the robot pose through our SDK at max 250Hz, or interpret report channel "REPORT_TCP_DEVELOP"(developer manual page 65) which updates at 100Hz.

Normally our planner in control box will help do trajectory planning to specified target. If you would like to control xArm position in real-time, please refer [here](https://github.com/xArm-Developer/xarm_ros/tree/master/examples#2-servo_cartesian-streamed-cartesian-trajectory) for "servo_cartesian" and "servo_joint" interface, which support max command frequency at 250Hz theoretically.

Velocity control will also come soon.

JuliusSustarevas, 2022-01-19:
Hi @penglongxiang Could you elaborate a bit what would I need to do to get higher joint state out of the arm? Maybe point to specific lines/files?

JuliusSustarevas, 2022-01-19:
In hardware interface I see line 92 is now commented out:
```// pos_sub_ = root_nh.subscribe(jnt_state_topic, 100, &XArmHW::pos_fb_cb, this);
Which i guess its good you no longer just republish states that were published using the driver. 

I see that read is back and is actually calling
read_code_ = xarm.getServoAngle(curr_read_angles_);

 does this mean from firmware 1.8 joint state  feedback is actually faster? And it would be regulated by the frequency of joint_state_controller?

penglongxiang, 2022-01-21:
Hi @JuliusSustarevas , we added the read() in hardware interface for better reaction when communication fails. Currently the ros service method did not show quite significant influence yet, and we've set the frequently called services to be persistent service. But indeed, this may not be a good implementation, we will consider to remove the service calling and potential delay, thanks for your kind advice.

JuliusSustarevas, 2022-01-21:
Hi but can you confirm that reading directly at say 100hz is ok? Using the
getservoangle function? from the driver?

JuliusSustarevas, 2022-01-21:
I am using a commit from july 2021 and don't want to update just because
everything seems to work so.. no reason to introduce new things :D I just
wanna get that joint_states callback. In the commit I'm using  joint_states
is coming from a report callback. So I plan to just fork and change that to
use the getservoangle

penglongxiang, 2022-01-24:
According to our recent test, reading (by xArmHW) at 100Hz seems to be OK. As long as the network connection is reliable and stable.