xArm-Developer/xarm_ros #56: sleep_sec ROS topic

swiz23, 2020-12-23:
What does this ROS topic do?

penglongxiang, 2020-12-24:
Hi, this is to tell the controller to sleep for some time (in seconds) before executing next command.

swiz23, 2020-12-24:
So you publish to the topic once in the beginning of your program to tell the controller how long to sleep between any number of future commands? Or you publish to the topic after any command you want it to sleep at?

What's the advantage of this over rospy.sleep(), time.sleep(), or the cpp equivalent?

penglongxiang, 2020-12-25:
> So you publish to the topic once in the beginning of your program to tell the controller how long to sleep between any number of future commands? Or you publish to the topic after any command you [...]

It is the later case. Use this topic only where sleep is needed. The advantage would be perhaps its accuracy, sometimes the sleep() function in user PC may get pre-empted by OS (especially Windows) and have unpredictable delay. This sleep instruction is performed inside controller box's real-time system.