xArm-Developer/xarm_ros #57: Rviz doesn't show the planned trajectory (trace)

AnsunWu, 2021-01-03:
Hello,

Recently, I learned the package of MoveIt! and try to use X-arm to do some simulations.
I met with a problem: **RVIZ doesn't show the planned trajectory.**

The code in the "xarm_simple_planner.CPP" in the x-arm planner_package, the "rviz_visul_tool" is already being used to show the information and the planned trajectory in Rviz. (Shown below)

[image]
[image]

And I also added the "Marker Array" tool in Rviz and subscribe to the right topic, the information is shown correctly:
[image]

But when I executed the "pose planning code", the robot trajectory was now shown in Rviz as the code...
`rosservice call xarm_pose_plan 'target: [[0.28, 0.2, 0.2], [1.0, 0.0, 0.0, 0.0]]'
`
[image]

**Could you please tell me how to let the rviz show the trajectory of the robot as in the picture below?**
[image]

**I followed the detailed tutorial of the official moveit webpage and tried many ways to solve this problem, including **modifying the rviz environment, robot model in Moveit_setup_assistant,and modifying the code**...But still can not solve this problem.

Or is there any other way to show the trajectory like the picture above?
Many thanks for your help!!**

`http://docs.ros.org/en/kinetic/api/moveit_tutorials/html/doc/move_group_interface/move_group_interface_tutorial.html`

AnsunWu, 2021-01-04:
@penglongxiang Solve the problem, the source of the question is :  
 there is no end effector link in the URDF(XACRO) file of the X-arm 5,6,7, and the function of `publishTrajectoryLine` is aimed to show the trace of the end-effector link. That's why the planned trajectory is not shown in RVIZ.

Solving method: 
1. Modify the URDF file of the X-arm robot, add an end effector link into it and use a fixed joint to connect the link6 with the end-effector link.
2. Use the "Moveit_setup_assistant" tool, add a "plan group" of the end-effector, set end-effector, and generate the .srdf and  .yaml file.
3. Done.

penglongxiang, 2021-01-05:
@AnsunWu Thank you for providing the solution to this problem. Showing trajectory is originally intended but I did not achieve it successfully. We will fix this in next update.

AnsunWu, 2021-01-05:
> @AnsunWu Thank you for providing the solution to this problem. Showing trajectory is originally intended but I did not achieve it successfully. We will fix this in next update.

No problem, Hope to do my development with the new update :)
I will close the issue.

Thanks,谢谢。

closed 2021-01-05 by AnsunWu

penglongxiang, 2021-01-06:
Hi @AnsunWu, we have fixed the trajectory visualization issue, please refer [here](https://github.com/xArm-Developer/xarm_ros/tree/master/xarm_planner) and update your local repository, you can now use xarm models with UF gripper or vacuum gripper attached in 'xarm_planner'.

AnsunWu, 2021-01-06:
@penglongxiang Hello, Thanks for the quick update!