xArm-Developer/xarm_ros #59: xarm_driver node crashing

naktamello, 2021-02-02:
Hello, I would like to report a strange behavior while using xarm7.  This happens quite often for me.

xarm model: xarm7 with DC PSU
ROS version: melodic
using with moveit: yes
host computer: Jetson Xavier AGX

### behavior:
1) xarm_driver crashes while executing motion with moveit, sometimes with sudden abrupt movements.

2) sometimes error happens even while stationary, like right after starting the node.  Also if I leave ROS running and come back after lunch break, it would have crashed similarly. However robot never crashes or moves abruptly while using xArm studio.

### Please see the following ROS log errors:
1) while executing moveit trajectory, sudden abrupt movements occur and these errors are printed in rosconsole:
```
Error: real_data.flush_data failed, ret = -2
Error: real_data.flush_data failed, ret = -2
Error: real_data.flush_data failed, ret = -2
[ WARN] [1611033990.500463346]: joint 3 abnormal command! previous: -0.801784, this: -1.337893

[ WARN] [1611033990.501630547]: joint 4 abnormal command! previous: 0.463958, this: 0.393108

[ WARN] [1611033990.502213124]: joint 5 abnormal command! previous: 0.670980, this: 1.119614

[ WARN] [1611033990.503170286]: joint 6 abnormal command! previous: 1.059416, this: 0.517504

[ WARN] [1611033990.506442686]: joint 7 abnormal command! previous: -1.413036, this: -2.357749

SocketPort::recv_proc exit, 11
[ERROR] [1611033994.255229374]: Failed to call service move_servoj
[ERROR] [1611033994.255556321]: Failed to call service move_servoj
[ERROR] [1611033994.266150062]: Failed to call service move_servoj
[ERROR] [1611033994.275791115]: Failed to call service move_servoj
[xarm/xarm_driver-2] process has died [pid 19393, exit code -11, cmd /home/bgoat/catkin_ws/devel/lib/xarm_api/xarm_driver_node __name:=xarm_driver __log:=/mnt/1005750b-a346-4aa9-af8a-a586d9ec3c3e/.ros/log/b6316258-5a16-11eb-9225-48b02d2b82fa/xarm-xarm_driver-2.log].
log file: /mnt/1005750b-a346-4aa9-af8a-a586d9ec3c3e/.ros/log/b6316258-5a16-11eb-9225-48b02d2b82fa/xarm-xarm_driver-2*.log
[ERROR] [1611033994.285811458]: Failed to call service move_servoj
[ERROR] [1611033994.296369429]: Failed to call service move_servoj
[ERROR] [1611033994.305735174]: Failed to call service move_servoj
[ERROR] [1611033994.315706662]: Failed to call service move_servoj
```

2) same behavior as above but node does not exit, instead prints `error: socket_send_data` endlessly.
```
[ERROR] [1612074783.132805788]: 
Invalid Trajectory: start point deviates from current robot state more than 0.01
joint 'joint1': expected: -3.1056, current: -1.39891
[ INFO] [1612074783.133137961]: Execution completed: ABORTED
SocketPort::recv_proc exit, 12
Error: real_data.flush_data failed, ret = -2
[ WARN] [1612074783.972975479]: joint 1 abnormal command! previous: -1.736944, this: -1.791387

[ WARN] [1612074783.973332710]: joint 2 abnormal command! previous: -0.780556, this: -0.927827

[ WARN] [1612074783.973896477]: joint 3 abnormal command! previous: 0.648341, this: 0.735660

[ WARN] [1612074783.974116902]: joint 4 abnormal command! previous: 0.317356, this: 0.515447

[ WARN] [1612074783.974246635]: joint 5 abnormal command! previous: 1.585996, this: 1.660190

[ WARN] [1612074783.974344143]: joint 6 abnormal command! previous: 0.623723, this: 0.940028

[ WARN] [1612074783.974489973]: joint 7 abnormal command! previous: -0.068890, this: -0.274293

[net work] error: socket_send_data
[net work] error: socket_send_data
[net work] error: socket_send_data
[net work] error: socket_send_data
[net work] error: socket_send_data
[net work] error: socket_send_data
[net work] error: socket_send_data
[net work] error: socket_send_data
[net work] error: socket_send_data
[net work] error: socket_send_data
[net work] error: socket_send_data
[net work] error: socket_send_data
```

3) the following error is printed right after nodes are started and I am unable to move the robot.
```
[ INFO] [1612074897.322016087]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1612074897.322227072]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1612074897.886367954]: Recovery behavior will clear layer 'obstacles'
[ INFO] [1612074897.907037753]: Recovery behavior will clear layer 'obstacles'
[INFO] [1612074898.013211]: Started controllers: xarm7_traj_controller
[ INFO] [1612074898.049392203]: odom received!
Error: real_data.flush_data failed, ret = -2
Error: real_data.flush_data failed, ret = -2
sizeof_data = 69
Error: real_data.flush_data failed, ret = -1
[ WARN] [1612074904.602190357]: Could not grab single echo scan.
[ WARN] [1612074904.728127084]: Hardware Notification:RT IC2 Config error,1.61207e+12,Error,Hardware Error
[ERROR] [1612074904.729105268]: Performing Hardware Reset.
sizeof_data = 69
Error: real_data.flush_data failed, ret = -1
[ WARN] [1612074915.162202866]: Could not grab single echo scan.
Error: real_data.flush_data failed, ret = -2
sizeof_data = 69
Error: real_data.flush_data failed, ret = -1
[ WARN] [1612074925.982250721]: Could not grab single echo scan.
Error: real_data.flush_data failed, ret = -2
Error: real_data.flush_data failed, ret = -2
Error: real_data.flush_data failed, ret = -2
```

I will do my best to provide more information if needed. Thanks,

penglongxiang, 2021-02-02:
Hi @naktamello , could you provide more information: 1) controller firmware version (shown in xArm studio).  2) the estimated date of your last  'xarm_ros' update. 3) the network connection method between your host PC and xArm controller box (wired or wireless, with additional relays/routers or not). 4) Did you run xArm studio on the same host computer you mentioned?   

It seems the data transmission to your host computer (running xarm_driver) is not stable.

naktamello, 2021-02-02:
1) fw version is 1.6.5
2) xarm_ros:
```
commit 87c08a520cf0c17c10676d12aa897c7c8b237c5e (HEAD -> master, origin/master, origin/HEAD)
Author: Jason Peng <penglongxiang1988@gmail.com>
Date:   Wed Dec 23 10:58:30 2020 +0800
```
3) It is wired connection with an ethernet switch between xArm and host computer. No other devices are connected in the switch, and cables are < 30cm.
4) xArm studio was run on a different computer (because it is not supported on arm64 architecture)

So it seems I should give it a shot without the ethernet switch right? I will get back to you after I try.

Do you think it is possible to recover from socket error when it happens by modifying the xarm_driver?

penglongxiang, 2021-02-02:
@naktamello Thanks for your information. When xArm Moveit trajectory controller is running on host computer, it sends command to xArm in 100Hz. So a stable connection must be ensured, you can test round-trip time (RTT) by `ping <controller-ip>`, it is important to keep  RTT ***within 10ms***.  

Moreover, it's not recommended to run CPU or IO consuming applications alongside with Moveit real-robot controller, which would introduce more delay to data transmission to xArm control box. If you are running such applications, please stop them first and try again. 

I will see if anything can be done to improve the performance in unstable network.

naktamello, 2021-02-07:
First I removed ethernet switch between Jetson and xarm controller.
While my full launch file is running, I pinged the controller and got RTT < 1ms:
`rtt min/avg/max/mdev = 0.232/0.277/0.323/0.032 ms`

However as you guessed I am using a lot of CPU heavy nodes along with xarm_driver.
Sometimes I can go for 30~40 minutes without crashing and sometimes it would happen right away.

I noticed you just added print message to `socket.cc`. I will check if any helpful messages are printed.

naktamello, 2021-02-11:
@penglongxiang,
Here is a screenshot of the error with the new print message.
FYI, 'urg_node' is a node running hokuyo lidar connected via USB2.0-to-ethernet converter.

It will be awesome if xarm_driver can recover from such errors without having to restart the node.
Thanks and happy CNY. Hope to hear from you after the holidays.

[image]

penglongxiang, 2021-02-12:
@naktamello Thanks for your detailed feedback, I will analyse the cause of this phenomenon and try to fix this issue after we are back to office a week later. Also wish you a happy and lucky new year.

penglongxiang, 2021-03-02:
Hi @naktamello , it may require a **controller firmware update** to improve the connection in unstable network, this update is planned to be released in **Late March**.

naktamello, 2021-03-02:
@penglongxiang, thanks for your effort to resolve the issue. I'm looking forward to get the fw update!

Meanwhile I opened up the DC control box and tried bypassing the RJ45 extension cord from controller computer to the controller metal frame (DC version), and it does help reduce frequency of the error. I'll have to run it more to be absolutely sure, but the extension cord seems to have a negative impact on communication quality.

If you need beta testing of new fw lmk, I'll be glad to test and report back.

naktamello, 2021-04-11:
This issue is fixed after uploading new controller fw AND updating to latest version of xarm_ros (1f9c75dc509ab63b7877202d8a0964ba48107594, as of today)
Thank you for working on this problem, @penglongxiang!

closed 2021-04-11 by naktamello