xArm-Developer/xarm_ros #61: Driver sends robot commands when controllers fail to load (!)

felixvd, 2021-03-03:
When starting up the robot driver before the `robot_description` parameter is set, it reports an error and outputs that no controllers were started up, **but then sends zero-position commands to all joints!** This is dangerous behavior.

You can reproduce the behavior by starting this launch file on a fresh roscore (you may also need a real robot). Commenting out the line that uploads `robot_description` causes the bug to occur. 
```
<?xml version="1.0"?>
<launch>
  <arg name="robot_ip" default="192.168.1.214"/>

  <!-- If this line is removed, the driver sends joint commands upon startup -->
  <param name="robot_description" command="xacro --inorder  '$(find xarm_description)/urdf/xarm6_robot.urdf.xacro' limited:=true"/>

  <remap from="/xarm/joint_states" to="/joint_states"/>
  <include file="$(find xarm_bringup)/launch/xarm6_server.launch">
    <arg name="robot_ip" value="$(arg robot_ip)" />
    <arg name="use_moveit" value="true" />
    <arg name="ns" value="xarm" />
  </include>
  
  <rosparam file="$(find xarm_controller)/config/xarm6_traj_controller.yaml" command="load"/>
  <node
    name="traj_controller_spawner"
    pkg="controller_manager"
    type="spawner"
    respawn="false"
    output="screen"
    ns="xarm"
    args="xarm6_traj_controller"/>
```

Here is the terminal output when no `robot_description` is found:

```
process[xarm/traj_controller_spawner-1]: started with pid [59136]
process[xarm/xarm_driver-2]: started with pid [59137]
process[xarm/xarm_traj_controller-3]: started with pid [59138]
[ INFO] [1614739172.494982512]: start xarm driver
Tcp Report Norm connection successful
Tcp Control connection successful
[INFO] [1614739172.724230]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1614739174.742081852]: motion enable, ret = 0

[ INFO] [1614739174.756223941]: servo mode, ret = 0

[ INFO] [1614739174.772531233]: start, ret = 0

[INFO] [1614739174.832211]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1614739174.834114]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1614739174.835735]: Loading controller: xarm6_traj_controller
[ERROR] [1614739174.841896763]: Could not find parameter robot_description on parameter server
[ERROR] [1614739174.841966595]: Failed to parse URDF contained in 'robot_description' parameter
[ERROR] [1614739174.841986839]: Failed to initialize the controller
[ERROR] [1614739174.842015245]: Initializing controller 'xarm6_traj_controller' failed
[ERROR] [1614739175.843300]: Failed to load xarm6_traj_controller
[INFO] [1614739175.844406]: Controller Spawner: Loaded controllers: 
[INFO] [1614739176.794854]: Started controllers: 
```

When `robot_description` is available, the driver starts up healthy and sends no joint commands:

```
process[xarm/xarm_driver-1]: started with pid [60294]
process[xarm/xarm_traj_controller-2]: started with pid [60295]
[ INFO] [1614740763.971844821]: start xarm driver
process[xarm/traj_controller_spawner-3]: started with pid [60300]
Tcp Report Norm connection successful
Tcp Control connection successful
[INFO] [1614740764.220008]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1614740766.219007836]: motion enable, ret = 0

[ INFO] [1614740766.233351613]: servo mode, ret = 0

[ INFO] [1614740766.249698320]: start, ret = 0

[INFO] [1614740766.327826]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1614740766.329563]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1614740766.331418]: Loading controller: xarm6_traj_controller
[INFO] [1614740766.350450]: Controller Spawner: Loaded controllers: xarm6_traj_controller
[INFO] [1614740768.270390]: Started controllers: xarm6_traj_controller
```

penglongxiang, 2021-03-03:
Hi @felixvd,

Could you show the full text of your version of application launch file? It seems to be unfinished.

It is not recommend for general user to modify the content order and remove components from the provided launch files. Unless you have tracked all the linked code and know exactly what your modification can cause.

I noticed your provided text is similar to  "realMove_exec.launch" in xarm6_moveit_config dir, but missing the third main part. In this file "**xarm6_traj_controller**" is meant to receive command from **move_group** node through "follow_joint_trajectory" action topic. If move_group node and its **planning context** (robot_description needed) are not properly configured and started, the controller may send 0 command to robot. 

We will see what can be done to prevent this situation from happening after all.

felixvd, 2021-03-03:
I added the missing `</launch>` to the launch file. Excuse me, I was in a hurry.

> I noticed your provided text is similar to "realMove_exec.launch" in xarm6_moveit_config dir, but missing the third main part. In this file "**xarm6_traj_controller**" is meant to receive command [...]

Yes, I took the relevant (non-MoveIt) parts of `realMove_exec.launch` to start up the robot controller separate from MoveIt. It is my understanding that this is a standard design pattern, and that it is best practice to separate the robot driver from the MoveIt startup, so that the physical robot can be replaced with a simulation at the controller interface.

I understand that this is out of the order of the ReadMe, but I think you agree that it is unintuitive that a robot driver would send commands to the robot, although it has not received any input. Especially if it has reported an error upon startup.

Inserting the `robot_description` parameter into the launch file fixed the problem for us, so I can't spend much time debugging this, but I suspect that there will be a suitable point in `xarm_driver` or `xarm_traj_controller` to check if initialization has failed, and raise an exception instead of continuing to start up the driver. There is already an error message – maybe killing the node at this point is sufficient.

At the very least I would add a comment to the launch file, but I believe it would be much safer if you don't let the driver send any commands at all, until it is certain that the controllers are set up correctly.

penglongxiang, 2021-03-03:
@felixvd Agreed. It did not occur to us that this situation may happen. Thanks for pointing out the issue. Will try to add the error checking in next updates.