xArm-Developer/xarm_ros #62: Accessing IOState

micahpearlman, 2021-03-04:
Is there a method to access `IOState` while using the ROS API?  Doing research I found the following code commented out in `xarm_driver_node.cpp`:

```
// publish io state: This line may interfere with servoj execution
// xarm_driver.pub_io_state();
```

What specifically we are trying to do is to detect *BOTH* that the stop button is pressed or released.  Currently we can detect if the stop button is pressed via the `/xarm/xarm_states` topic and checking `if robot_msg.state == 4` but we cannot determine if the button is physically in the "up" or "ready" position.  

One idea was to assign the "Emergency Stop" function to a digital pin and monitor the digital pin, but in ROS there seems to be no way to do this without getting and publishing the `IOState` message.  

We also tried to do this by using the C++ SDK `get_cgpio_state` in a ROS node but that seems to cause issues with the ROS driver.

penglongxiang, 2021-03-04:
Please refer to [xArm controller IO ROS service instructions](https://github.com/xArm-Developer/xarm_ros#controller-io-operations), I believe you can use `/xarm/get_controller_din` service and check the target controller IO port (`io_num`) periodically. Please note the `io_num` for inquiry is from 1 to 8 for corresponding CI0 to CI7 marked at controller side.

micahpearlman, 2021-03-04:
The `/xarm/get_controller_din` does not appear to respond when the emergency stop button is released.

The procedure to reproduce:

1. In xArm Studio set the IO output function to "Emergency Stop": 
[image]
2. With system running and with emergency stop button up check `/xarm/get_controller_din`:
```
root@ccc35493cc64:/catkin_ws# rosservice call /xarm/get_controller_din "io_num: 1" 
ret: 0
value: 1
message: "get Controller digital Input ret = 0"
```
3.  Press the emergency stop button and check `/xarm/get_controller_din`:
```
root@ccc35493cc64:/catkin_ws# rosservice call /xarm/get_controller_din "io_num: 1" 
ret: 1
value: 1
message: "get Controller digital Input ret = 1"
```
4. Now reset the emergency stop button back to up position and check the `/xarm/get_controller_din`:
```
root@ccc35493cc64:/catkin_ws# rosservice call /xarm/get_controller_din "io_num: 1" 
ret: 1
value: 1
message: "get Controller digital Input ret = 1"
```
5. Note that the digital input return value stays at `1` even though the emergency stop button is back in the up position.  We want to be able to detect if the emergency stop button is in the up position after it has been in the down position.

micahpearlman, 2021-03-04:
Also note if we use the C++ SDK example `5008-get_cgpio_state` and have the  IO output function as above and run the following procedure:

1. With the emergency button up and in normal running condition:
```
root@ccc35493cc64:/xarm_cpp_sdk/build/example# ./5008-get_cgpio_state 192.168.1.234
SDK_VERSION: 1.6.0
Tcp control connection successful
is_old_protocol: 0
version_number: 1.6.0
Tcp report connection successful
[set_state], xArm is ready to move
=========================================
get_cgpio_state, ret=0
* state=0, err_code: 0
* digital input functional gpio state=256
* digital input configuring gpio state=65535
* digital output functional gpio state=0
* digital output configuring gpio state=65280
* analog-0 input val=0.043956
* analog-1 input val=0.053724
* analog-0 output val=0.000000
* analog-1 output val=0.000000
* digital input functional info, io0=0, io1=0, io2=0, io3=0, io4=0, io5=0, io6=0, io7=0
* digital output functional info, io0=1, io1=1, io2=15, io3=0, io4=15, io5=0, io6=15, io7=0
```
2. Press the emergency stop button:
```
root@ccc35493cc64:/xarm_cpp_sdk/build/example# ./5008-get_cgpio_state 192.168.1.234
SDK_VERSION: 1.6.0
Tcp control connection successful
is_old_protocol: 0
version_number: 1.6.0
Tcp report connection successful
[set_state], xArm is ready to move
=========================================
get_cgpio_state, ret=0
* state=0, err_code: 0
* digital input functional gpio state=256
* digital input configuring gpio state=65535
* digital output functional gpio state=0
* digital output configuring gpio state=65280
* analog-0 input val=0.043956
* analog-1 input val=0.053724
* analog-0 output val=0.000000
* analog-1 output val=0.000000
* digital input functional info, io0=0, io1=0, io2=0, io3=0, io4=0, io5=0, io6=0, io7=0
* digital output functional info, io0=1, io1=1, io2=15, io3=0, io4=15, io5=0, io6=15, io7=0
```
3. Release the emergency stop button to up position:
```
root@ccc35493cc64:/xarm_cpp_sdk/build/example# ./5008-get_cgpio_state 192.168.1.234
SDK_VERSION: 1.6.0
Tcp control connection successful
is_old_protocol: 0
version_number: 1.6.0
Tcp report connection successful
[set_state], xArm is ready to move
=========================================
get_cgpio_state, ret=0
* state=0, err_code: 0
* digital input functional gpio state=256
* digital input configuring gpio state=65535
* digital output functional gpio state=0
* digital output configuring gpio state=65280
* analog-0 input val=0.039072
* analog-1 input val=0.051282
* analog-0 output val=0.000000
* analog-1 output val=0.000000
* digital input functional info, io0=0, io1=0, io2=0, io3=0, io4=0, io5=0, io6=0, io7=0
* digital output functional info, io0=1, io1=1, io2=15, io3=0, io4=15, io5=0, io6=15, io7=0
```
4. Note that `digital output configuring gpio state` changes from `65280` to `65283` and back to `65280` indicating that digital out `CO0` and `CO1` get switched as expected.  Unfortunately using the C++ SDK externally (like as above) or within a ROS node to check this seems to freeze our ROS setup.  Is the C++ SDK is not designed to work with the ROS SDK?

penglongxiang, 2021-03-05:
After setting the output signal in xArm studio, you may need to connect CO0/CO1 with a controller Input port by a physical wire, in this way the "/xarm/get_controller_din" will properly read the output state. And you should read the "value" field instead of "ret" in service response.

Unfortunately, **the Output signal will not change if Emergency button is released from pressed status**, unless motor is successfully enabled and robot state is set to 0. Currently, this  "Emergency Stop" output signal will be triggered both by hardware and software. 

We have noticed your feedback and are now re-evaluating this signal logic. You may try the following method at this moment:
1. call the service "`/xarm/get_err`" with no argument, if you get `err: 1` which means the power is cutoff and emergency button was pressed previously. 
2. under this situation (err = 1), call service "`/xarm/set_state  0`"  if you got "`ret: 0`", the button has been released, you can re-enable the motors to continue. If `ret: 1` the button has not been released yet.

No need for the IO signal which currently can not tell if emergency button has been released.

micahpearlman, 2021-03-12:
Thank you that work as we have hoped! (calling the `get_err` and `set_state`)

closed 2021-03-12 by micahpearlman