xArm-Developer/xarm_ros #63: Manual control using joystick with moveit

RexDrac, 2021-03-11:
Hi,

I am currently trying to control the xarm6 with a joystick.

The solution proposed in issue 16 is to use "Servo_Cartesian" and calling "/xarm/move_servo_cart".
[https://github.com/xArm-Developer/xarm_ros/issues/16](url)

A problem with this method is that the "move_servo_cart" can not be used concurrently with moveit.
Is there a way to achieve manual control using joystick with moveit?

Thanks

penglongxiang, 2021-03-12:
After the new firmware version is released in late March (hopefully), you can try the new velocity mode to work with joystick. There will be two new modes available: joint velocity control and Cartesian velocity control.

lorepieri8, 2021-09-07:
@RexDrac @penglongxiang  Has this been solved?
I'm using both the Python API and moveit. If I  send a cartesian command with set_servo_cartesian the arm executes the trajectory, but then quickly go back to the initial position. This doesn't happen is moveit is off, and I just use the Python API.

I've tried using moveit with velocity control,:

`roslaunch xarm6_nyrvan_gripper_moveit_config realMove_exec.launch robot_ip:=myip velocity_control:=true`

but this doesn't seem to solve the issue. 

I'm using set_mode(1) in the Python code. Firmware 1.7.2.

#34 may be related, even though I'm not sending any switching mode command during motion.

lorepieri8, 2021-09-07:
I just tested with set_mode(4)  and set_mode(5),  using respectively vc_set_cartesian_velocity and vc_set_joint_velocity. There is no interference with moveit.

Still, it would be good know if there is a way of using set_servo_cartesian and moveit together.

penglongxiang, 2021-09-09:
hi @lorepieri8 , as stated in [this section](https://github.com/xArm-Developer/xarm_ros/tree/master/examples#2-servo_cartesian-streamed-cartesian-trajectory), Moveit (position control mode) should not be used together with `set_servo_cartesian` or `set_servo_joint` because they operate in the same mode(1) and can interfere with each other. With `/xarm/xarm_traj_controller` running, it will keep sending servo command no matter there is new trajectory command from move_group or not.  

`vc_set_cartesian_velocity` will not interfere because it operates in velocity mode(5). However, if moveit and hardware interface are configured as velocity interface. They will also interfere.

In your case, the only way to try may be making xArmHW::need_reset()  return `true` during joystick operation. As what happens when mode changes, STOP state(4) or error occurs. It is called in xarm_control_node.cpp, refer to:  
https://github.com/xArm-Developer/xarm_ros/blob/004c5cd5971aed53c86a4af7e1047e59a085e365/xarm_controller/src/xarm_hw.cpp#L348 

https://github.com/xArm-Developer/xarm_ros/blob/004c5cd5971aed53c86a4af7e1047e59a085e365/xarm_controller/src/xarm_control_node.cpp#L37  

It will reset the running trajectory controller to use current joint state as new command.

ShawnSchaerer, 2022-02-22:
You can also integrate Moveit servo
https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html

Right now I have this working with gazebo XARM 7 package and can switch between moveit and sending tip velocities.
The simulation is a bit jerky but I think its due to the PID tuning in gazebo.  I will be executing this on the real XARM 7 to be certain.

You can use both Moveit servo and moveit together, but you will have to switch between controllers at the command line or using a custom UI.