xArm-Developer/xarm_ros #64: Limited explanation in docs

JuliusSustarevas, 2021-03-17:
Could you add to the xarm_description readme. what the "limited"  specification does? 
I see that default xarm6 limits are:
`  <xacro:macro name="xarm6_urdf" params="prefix
    joint1_lower_limit:=${-2.0*pi}  joint1_upper_limit:=${2.0*pi}
    joint2_lower_limit:=${-2.059}  joint2_upper_limit:=${2.0944}
    joint3_lower_limit:=${-3.927}  joint3_upper_limit:=${0.19198}
    joint4_lower_limit:=${-2.0*pi}  joint4_upper_limit:=${2.0*pi}
    joint5_lower_limit:=${-1.69297}  joint5_upper_limit:=${pi}
    joint6_lower_limit:=${-2.0*pi}  joint6_upper_limit:=${2.0*pi}">`

While if you select limited they would be:

`   <xacro:xarm6_urdf prefix="${prefix}"
        joint1_lower_limit="${-pi*0.99}" joint1_upper_limit="${pi*0.99}"
        joint2_lower_limit="${-2.059}" joint2_upper_limit="${2.0944}"
        joint3_lower_limit="${-pi*0.99}" joint3_upper_limit="${0.19198}"
        joint4_lower_limit="${-pi*0.99}" joint4_upper_limit="${pi*0.99}"
        joint5_lower_limit="${-1.69297}" joint5_upper_limit="${pi*0.99}"
        joint6_lower_limit="${-pi*0.99}" joint6_upper_limit="${pi*0.99}"/>`

I see its more strict. but whats the usecase?

penglongxiang, 2021-03-18:
@JuliusSustarevas "limited" parameter is mainly for Moveit application usecase. If joint range is beyond [-pi, pi], Moveit tend to generate trajectories with greater joint motions, which for most users is unwanted. We will add some explanation later in the ReadMe.

goyalvarun02, 2022-08-12:
Hi @penglongxiang 

a) When to use limited:=false , when not using move it? please confirm.

b) If using limited:= true, move it plan the trajectory within the scope and in range [-pi, pi], In this is it also checking that if any singularity and self collision occur it will fail at planning stage or execution stage? (I suspect at planning stage it will failed).
 
c) What is the requirement of checking fraction <0.9 in function do_cartesian_path() function in file  xarm_simple_planner.cpp ?

Thanks

penglongxiang, 2022-08-22:
hi @goyalvarun02 ,

a) `limited:=false` can be always effective unless you really need some of the joints to go beyond +/- 180 degree, it mainly can make the iterative Inverse Kinematics used by Moveit generate unpredictable results.

b) If singularity or collision occurs, I believe it will fail at planning stage in Moveit.

c) This is to make sure a reasonable coverage of the planned path before considering it as valid. You can actually make it 100% if necessary.