xArm-Developer/xarm_ros #70: Issues installing with colcon

dignakov, 2021-06-23:
I'm using `colcon` to build ROS1 packages.

When installing `xarm_ros` with `colcon build` I get the following error:

```
CMake Error at /home/dmitri/catkin/ws_xarm/install/xarm_sdk/share/xarm_sdk/cmake/xarm_sdkConfig.cmake:110 (message):
  Project 'xarm_sdk' specifies 'include' as an include dir, which is not
  found.  It does not exist in
  '/home/dmitri/catkin/ws_xarm/install/xarm_sdk/include'.  Report the problem to
  the maintainer 'vinman <vinman.cub@gmail.com>' and request to fix the
  problem.
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  CMakeLists.txt:12 (find_package)
```

It looks like the CMake file for `xarm_sdk` doesn't install it's include directory, perhaps it should?

Any advice would be very helpful.

penglongxiang, 2021-06-24:
Our ROS1 packages here are targeted and tested under 'catkin' build system. Building with colcon may not work, CMakeLists are not the same. May I ask the reason for using colcon (which is used by ROS2)?

dignakov, 2021-06-24:
colcon can also be used for ROS1, and it results in an install space that can be deployed/copied. So using catkin is not an option on my end...

I spent some time looking at making it work, and I think there would need to be quite a few changes, but if you're open to it I'll submit a PR...

You also have a ROS2 package, which looks like it defines the proper install targets...

dignakov, 2021-06-24:
It also seems like `catkin_make_isolated` is also failing when it goes to build `xarm_api` it can't find `xarm_cxx_sdk`

```
==> Processing catkin package: 'xarm_api'
==> Creating build directory: 'build_isolated/xarm_api'
==> Building with env: '/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_sdk/env.sh'
==> cmake /home/dmitri/catkin/ws_xarm/src/xarm_ros/xarm_api -DCATKIN_DEVEL_PREFIX=/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_api -DCMAKE_INSTALL_PREFIX=/home/dmitri/catkin/ws_xarm/install_isolated -G Unix Makefiles in '/home/dmitri/catkin/ws_xarm/build_isolated/xarm_api'
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/dmitri/catkin/ws_xarm/devel_isolated/xarm_api
-- Using CMAKE_PREFIX_PATH: /home/dmitri/catkin/ws_xarm/devel_isolated/xarm_sdk;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_planner;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_msgs;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_gazebo;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_device;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm7_vacuum_gripper_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm7_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm7_gripper_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm6_vacuum_gripper_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm6_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm6_gripper_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm5_vacuum_gripper_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm5_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm5_gripper_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/dual_xarm6_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_description;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_bringup;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm7_redundancy_res;/home/dmitri/catkin/ws_xarm/devel_isolated/multi_xarm5;/opt/ros/noetic
-- This workspace overlays: /home/dmitri/catkin/ws_xarm/devel_isolated/xarm_sdk;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_planner;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_msgs;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_gazebo;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_device;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm7_vacuum_gripper_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm7_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm7_gripper_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm6_vacuum_gripper_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm6_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm6_gripper_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm5_vacuum_gripper_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm5_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm5_gripper_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/dual_xarm6_moveit_config;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_description;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_bringup;/home/dmitri/catkin/ws_xarm/devel_isolated/xarm7_redundancy_res;/home/dmitri/catkin/ws_xarm/devel_isolated/multi_xarm5;/opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.5", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Found PY_em: /usr/lib/python3/dist-packages/em.py  
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/dmitri/catkin/ws_xarm/build_isolated/xarm_api/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.5") 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Configuring done
CMake Warning (dev) at CMakeLists.txt:46 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "xarm_cxx_sdk" of target "xarm_ros_client" does not
  exist.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at CMakeLists.txt:56 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "xarm_cxx_sdk" of target "xarm_driver_node" does not
[... cut ...]
make[2]: *** [CMakeFiles/move_test.dir/build.make:103: /home/dmitri/catkin/ws_xarm/devel_isolated/xarm_api/lib/xarm_api/move_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:292: CMakeFiles/move_test.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
<== Failed to process package 'xarm_api': 
  Command '['/home/dmitri/catkin/ws_xarm/devel_isolated/xarm_sdk/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /home/dmitri/catkin/ws_xarm/build_isolated/xarm_api && /home/dmitri/catkin/ws_xarm/devel_isolated/xarm_sdk/env.sh make -j4 -l4

Command failed, exiting.
```

penglongxiang, 2021-06-24:
Thank you for your feedback. We will look into this issue asap.

dignakov, 2021-06-24:
@penglongxiang I submitted a PR #71 which should fix the build. Please take a look.