xArm-Developer/xarm_ros #72: include directories should have a "package_name" folder

dignakov, 2021-06-24:
Currently the packages have their header files directly int he include directory.

Some of the file names (e.g. visibility_controller.h) and folder names (e.g. serial) are very general and having them on the include path directly can cause issues.

This would require some work, but I think it would be better to follow the structure of other ros packages and structire the include folder

package_name/include/package_name/<all of the headers>

The only issue to this is that xarm_sdk pacakge's external dependency would need some work as well.

I could submit a PR against both packages, if this is something you'd merge.

penglongxiang, 2021-06-30:
hi @dignakov, thanks for your feedback and suggestion. We managed to fix the build issues and follow the directory conventions in the latest commit (14ed624).

dignakov, 2021-06-30:
@penglongxiang Thanks, was anything in my PR useful?