xArm-Developer/xarm_ros #74: Procedure for setting up custom end-effector

JuliusSustarevas, 2021-07-03:
Hi, 

Apologies if I missed something and this already exists.  This question is related to https://github.com/xArm-Developer/xarm_ros/issues/49. 

I think this repo would benefit from a clear step by step on how to set up a custom end-effector on the xarm. 

My current understanding is as follows:
- xarm drivers/planners  have the "/xarm/set_tcp_offset" service  to change the end-effector frame for planning purposes. 
- This is persistent, i.e. between power cycles this offset remains stored in the control box (is this true?)
- also in xarm studio you can set the end-effector weight. Is this also persistent and remains over power cycles? 
- I believe/guess that the end-effector weight is used for internal dynamics calculations (i.e. joint effort) when detecting collisions? Is this true? 

Given this understand I wonder what are the right steps when setting up a custom end effector. My main concerns are that xarm drivers are aware of:
-  TCP offset. 
- end-effector weight (if its used for collision detection via joint torque.)
- and new end effector collision box. (TCP offset alone will not introduce a collision box, so we must add this to the description. However, since moveit movegroup is already created we cannot redefine the end-effector frame without rewriting the moveit config)

Could you please advise on what are the right "as intended" steps for setting up a custom end-effector? 

My current guess is:
- set TCP offset and end-effector weigh in xarm studio
- add your end-effector link with collision boxes to the description and attach to xarm flange appropriately. 
- do not use raw moveit because movegroup will still think flange is the end-effector. Use planner controllers instead - they will use the offset TCP, but since they call moveit - they will be aware of new collision boxes? 

I'm especially having trouble with the weight of the end-effector. The weight is only 1kg (I have xarm6) but the arm very often locks up and I'm not sure why. Is it weight setting? or should I reduce the sensitivity of collision detection?

penglongxiang, 2021-07-05:
Hi @JuliusSustarevas ,
 
> My current understanding is as follows:
>xarm drivers/planners have the "/xarm/set_tcp_offset" service to change the end-effector frame for planning purposes.   

This ”set_tcp_offset“ service is only effective for the **non-moveit** planning situation. This TCP position offset is seen by the planner inside xArm controller box, but not seen by Moveit which is a third party planner and not integrated with xArm controller system. That is, setting TCP offset or loading this way can only be effective for xArm studio/xArm SDK/xarm ros services provided by "xarm_driver_node" in xarm_api ros package.   

>This is persistent, i.e. between power cycles this offset remains stored in the control box (is this true?)  

This is true. If you called the service successfully.   

>also in xarm studio you can set the end-effector weight. Is this also persistent and remains over power cycles?  

Yes, same with tcp offset setting this way.  

>I believe/guess that the end-effector weight is used for internal dynamics calculations (i.e. joint effort) when detecting collisions?Is this true?  

Your guessing is correct.  

I suppose you want moveit to be aware of your custom end-effector. For tool position offset, "/xarm/set_tcp_offset" service or xArm studio configuration will not help, as explained above. You have to add the urdf description of your end-effector, attatch it to xArm6 robot description and add a fixed joint with the TCP offset. Then you will also need to modify the `xarm6.srdf` file in `xarm6_moveit_config/config` directory.  Adding additional links and joints, and update link collision settings accordingly.  

An example to follow is to understand how the vacuum gripper is attached by checking "`examples/xarm6_vacuum_gripper_moveit_config`".  We will do assessment on whether a demo or interface can be developed.  
 
As for weight setting, you can do it in xArm studio or calling "xarm/set_load" service. since the change of weight can be sensed directly by controller box. Please make sure the **Center of Mass** is also properly set other than the Mass, the parameters should be as precise as possible to prevent false triggering under high sensitivity.

JuliusSustarevas, 2021-07-05:
Thanks so much for detailed answer.  And again, apologies if I'm coming into this with a superficial understand of xarm planner vs moveit. 

One more question: 
- from what I see right now, your xarm planner and moveit pipelines were intended to be used quite separately, presumably for different use cases.  Dynamics and kinematics between both "stacks" can be setup mostly independently. However, one thing that the xarm_planner is missing is collision boxes? Is this true? Is it possible to for example control the arm into self collision using the  ros services like end-effector velocity control?

penglongxiang, 2021-07-06:
Hi @JuliusSustarevas just want to clarify "xarm_planner" package in this repository still uses Moveit for xArm motion planning, but it shows a programmatic way to implement. The ros-wrapped SDK motion services introduced [here](https://github.com/xArm-Developer/xarm_ros#57-xarm_apixarm_msgs) was planned inside our control box, which takes into account the TCP offset set through "xarm/set_tcp_offset" service.  

If you upgrade the firmware and xArm studio version to be v1.6.0 or later, the motions planned by our controller box will also check self-collision to prevent dagage, in order to make this function operate properly with end effector installed, it is important to set a tool collision model in "Settings->End Effector" section of xArm Studio.

penglongxiang, 2021-07-15:
Hi @JuliusSustarevas , we have developed a method for quick configuration of the custom end-effector for Moveit. But it may require **ROS Melodic or later** with its corresponding Moveit version to work. If you are interested, please check [here](https://github.com/xArm-Developer/xarm_ros#551-add-custom-tool-model-for-moveit),

closed 2022-01-19 by JuliusSustarevas