xArm-Developer/xarm_ros #76: Using grippers in Gazebo simulation

MinwooKim1202, 2021-07-10:
I want to use the gripper in Gazebo 

I think xarm_api is only available on the real xArm.

Please teach me the way.

Thank you for reading it.

penglongxiang, 2021-07-13:
hi, you can use Moveit to plan and control gripper as well in gazebo simulator. Refer [here](https://github.com/xArm-Developer/xarm_ros#to-run-moveit-motion-planner-along-with-gazebo-simulator), remember to launch with `add_gripper:=true`, the gripper is available as an independent planning group.

MinwooKim1202, 2021-07-13:
> hi, you can use Moveit to plan and control gripper as well in gazebo simulator. Refer [here](https://github.com/xArm-Developer/xarm_ros#to-run-moveit-motion-planner-along-with-gazebo-simulator), remember to launch with `add_gripper:=true`, the gripper is available as an independent planning group.

thank you!