xArm-Developer/xarm_ros #77: how to add camera link to the URDF

fujiehuang, 2021-07-16:
I have a D435 camera mounted on the camera stand. 
I need to add the camera link's pose to the URDF.
What values should I use for (x, y, z)?

fujiehuang, 2021-07-16:
what I'm looking for is similar to the following section, which is from the openmanipulator-x's URDF for D435

[https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_description/urdf/open_manipulator.urdf.xacro](url)

<joint name="camera_joint" type="fixed">
    <axis xyz="0 1 0" />
    <origin xyz="0.070 0.032 0.052" rpy="0 0 0"/> 
    <parent link="link5"/>
    <child link="camera_link"/>
  </joint>

vimior, 2021-07-20:
1. Download the d435.dae file and save it to the xarm_ros/xarm_description/meshes/other
2. Recompile (catkin_make)
3.  Run moveit example (specify parameters)
    `roslaunch xarm7_moveit_config demo.launch add_other_geometry:=true geometry_type:=mesh geometry_mesh_filename:=d435.dae geometry_mesh_origin_xyz:='"0.0149 0.0 0.0125"' geometry_mesh_origin_rpy:='"1.57079632679489 0 1.57079632679489"' geometry_mesh_tcp_xyz:='"0 0 0.025"'`

iskanderuse, 2021-12-15:
If You are creating your own robot configuration You can add Realsense standard way
```
<!-- Realsense D435 -->
  <xacro:arg name="use_nominal_extrinsics" default="false"/>
  <xacro:arg name="add_plug" default="true" />
  <xacro:arg name="use_mesh" default="true" />
  <xacro:include filename="$(find realsense2_description)/urdf/_d435.urdf.xacro" />
  
  <xacro:sensor_d435 parent="$(arg prefix)link_eef" 
            use_nominal_extrinsics="$(arg use_nominal_extrinsics)"
            name="camera_d435" 
            add_plug="$(arg add_plug)" use_mesh="$(arg use_mesh)">
    <origin xyz="0.08 0 0.012" rpy="0 ${-pi/2} ${pi}"/>
  </xacro:sensor_d435>
```

penglongxiang, 2021-12-16:
Now there is a more convenient way to attach realsense D435i with camera stand to xArm model. Refer to: https://github.com/xArm-Developer/xarm_ros#74-adding-realsense-d435i-model-to-simulated-xarm