xArm-Developer/xarm_ros #78: velocity_controllers/JointTrajectoryController on real hardware

JuliusSustarevas, 2021-07-20:
Hi again, 

I'm going through the package and trying to setup velocity_controllers/JointTrajectoryController controller with real hardware. Ofc, I load the URDF with velocity_control up and can confirm that robot description has VelocityJointInterface specified. However, I keep getting the following error; (don't mind that names, they are changed abit)

[ERROR] [1626773144.650255253]: Exception thrown: Could not find resource 'xarmjoint1' in 'hardware_interface::VelocityJointInterface'.
[ERROR] [1626773144.650295875]: Failed to initialize the controller
[ERROR] [1626773144.650325702]: Initializing controller 'xarm_joint_velocity_controller' failed

I'm beginning to suspect that your "xarm_control/XArmHW" hardware interface does not support  hardware_interface::VelocityJointInterface for joints?

JuliusSustarevas, 2021-07-20:
I think this might have something todo with this:

https://github.com/xArm-Developer/xarm_ros/blob/935828d1a438eff35a5380856d4aa9aca515f59d/xarm_controller/src/xarm_hw.cpp#L150

vimior, 2021-07-23:
The previous xarm_ros did not fully support speed mode control. If you need to use speed mode control, please refer to the following
1. Download the latest xarm_ros
2. Ensure that the firmware of the control box is up to date (1.7.0)
3. Specify the parameter velocity_control to be true when starting moveit
    `roslaunch xarm6_moveit_config realMove_exec.launch robot_ip:=192.168.1.239 velocity_control:=true`

JuliusSustarevas, 2021-07-26:
Just checked the last commit! it works, thanks!

closed 2021-07-26 by JuliusSustarevas