xArm-Developer/xarm_ros #79: Real Arm goes to zero position when launching twice

lorepieri8, 2021-07-20:
This looks related to #61 .

To reproduce (**please read all the post before trying this, it can be dangerous**), place the real arm at a non zero position, then type in two different shells:

`roslaunch xarm6_gripper_moveit_config realMove_exec.launch robot_ip:=192.168.1.208`

with the ip of your robot. The robot will (in my case very quickly) go to the zero position. 
**This is potentially very dangerous, since whatever is between the original arm position and the zero position will be hit.**

penglongxiang, 2021-07-21:
Thanks for your feedback. With the latest 'xarm_ros' and controller firmware, I did not reproduce the sudden returning home behavior, except when two launch commands were executed, this message kept popping up in the background:
```
[robot_state_publisher-7] restarting process
process[robot_state_publisher-7]: started with pid [16169]
[ WARN] [1626831486.403978542]: Shutdown request received.
[ WARN] [1626831486.404182648]: Reason given for shutdown: [new node registered with same name]
```
Could you please tell us the 'xarm_ros' version (by checking `git log`) and the controller firmware version (by checking xArm studio) and ROS version on your side? Thanks.

lorepieri8, 2021-07-21:
commit b5003fbd21e50f6e7f05f4ee1af2f42f95b11143   (6 Jan)
firmware 1.6.1
ros Noetic (rospy 1.15.11)

I will check if I can update, and in that case if the problem is still there.

lorepieri8, 2021-07-21:
I can confirm that this is solved in the latest xarm_ros version, even without updating the firmware.

closed 2021-07-21 by lorepieri8