xArm-Developer/xarm_ros #84: How to use inverse kinematics

Aki92-Xavier, 2021-09-13:
Hi, I wanted to use IK on Xarm6 but i couldn't find how to do it.
Now I'm using realsense D435 to get the [x,y,z] and move the arm to that target value.
Is there any IK script for Xarm6.

penglongxiang, 2021-09-13:
Hi, there is no ros-wrapper for IK yet. You can try our python or C++ APIs.  

For python API:
https://github.com/xArm-Developer/xArm-Python-SDK/blob/e8475781cb7365a3d4134fdda0ee359b6142ca8e/xarm/wrapper/xarm_api.py#L1633

A piece of sample code:
```
from xarm.wrapper import XArmAPI

ip = '192.168.1.227'
arm = XArmAPI(ip, is_radian=False)
coords = [439.7,189.7,322.8,75.1,-16,23.7]
print("target Coord: ")
print(coords)
# caution: returned solution may not always be the same 
for i in range(10):
	print("solution: %d"%(i+1,))
	code, solution = arm.get_inverse_kinematics(coords, input_is_radian=False, return_is_radian=False)
	print(code)
	if code:
		print("NO valid solution!")
		continue
	print(solution) 
```

Aki92-Xavier, 2021-09-13:
Thank you very much.
I would like to try using the APIs.
By the way where should  find the document ''code: See the API code documentation for details'' 
And can we try to use moveit! packages to IK the Xarm.

closed 2021-09-26 by Aki92-Xavier

jzx-gooner, 2021-12-14:
> Hi, I wanted to use IK on Xarm6 but i couldn't find how to do it. Now I'm using realsense D435 to get the [x,y,z] and move the arm to that target value. Is there any IK script for Xarm6.

Hi,
1. i use the zed camera to track a box and get the [x,y] in image and the distance [x,y,distance]
2. transform the [x,y,distance]  in zed camera coords  to  [x_arm,y_arm,z_arm] in arm coords. 
3. move the arm to [x_arm,y_arm,z_arm] to catck the box

I want to know ,in step2 ,how to make this transfrom and if i use the  D435,what is the magnitude of error in range 1m，can i get 1cm accurate. Thank you very much !

penglongxiang, 2021-12-14:
hi @jzx-gooner ,  after the camera calibration, you need to use `tf` module to publish the transform between robot tool frame and the camera image frame to the ROS system. Then you can lookup the transform from robot base link to the detected object. 

We are not expert with D435, please contact their official support for the performance questions.

jzx-gooner, 2022-01-10:
> hi @jzx-gooner , after the camera calibration, you need to use `tf` module to publish the transform between robot tool frame and the camera image frame to the ROS system. Then you can lookup the [...]
Hi.i could not use tf ,so i follow next 3 steps:
0.Based realsense D435  i get the [x,y,z] on the camera model
1.use the hand_eye_matrix to get [x1,y1,z1] on the tool model
2.use tool_base_link to robot_base_link matrix to get [x2,y2,z2] on teh base model,now we could control our robot to [x2,y2,z2] to grasp.

Is it ringht?
and how can i get the matrix between tool_base_link and robot_base_link.they are different in different position.

penglongxiang, 2022-01-11:
Hi @jzx-gooner , the 3 steps are correct. If you do not have access to tf or no urdf model passed to parameter server, to get the tool position relative to base for real xArm, first make sure xarm driver is properly started by `roslaunch xarm_bringup xarm7_server.launch robot_ip:=192.168.1.xxx` or equivalent. Then the TCP(tool center point) position can be acquired by:
```bash
rosservice call /xarm/get_position_rpy (orientation represented as roll-pitch-yaw)
```
or you can subscribe to topic `/xarm/xarm_states` and check the `pose` field. The message explanation and data unit (mm and radian) can be referred [here](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm_msgs/msg/RobotMsg.msg).