xArm-Developer/xarm_ros #85: Realsense D435 eye-in-hand

Aki92-Xavier, 2021-09-26:
Hi, I am using realsense D435 and Xarm6.
I found this launch file in xarm_ros/xarm_camera/launch/xarm_realsense_d435i_eyeinhand.launch.

Could you please explain how to use this launch file and does it calibrate?

<launch>
  <arg name="robot_ip" />
  <arg name="dof" default="7" />
  <arg name="report_type" default="normal" />

  <arg name="add_other_geometry" default="false"/>
  <arg name="geometry_type" default="box"/>
  <arg name="geometry_mass" default="0.1"/>
  <arg name="geometry_height" default="0.1"/>
  <arg name="geometry_radius" default="0.1"/>
  <arg name="geometry_length" default="0.1"/>
  <arg name="geometry_width" default="0.1"/>
  <arg name="geometry_mesh_filename" default=""/>
  <arg name="geometry_mesh_origin_xyz" default="'0 0 0'"/>
  <arg name="geometry_mesh_origin_rpy" default="'0 0 0'"/>
  <arg name="geometry_mesh_tcp_xyz" default="'0 0 0'"/>
  <arg name="geometry_mesh_tcp_rpy" default="'0 0 0'"/>

  <arg if="$(eval arg('dof') == 5)" name="moveit_launch_file" value="$(find xarm5_moveit_config)/launch/realMove_exec.launch" />
  <arg if="$(eval arg('dof') == 6)" name="moveit_launch_file" value="$(find xarm6_moveit_config)/launch/realMove_exec.launch" />
  <arg if="$(eval arg('dof') == 7)" name="moveit_launch_file" value="$(find xarm7_moveit_config)/launch/realMove_exec.launch" />

  <include file="$(arg moveit_launch_file)">
    <arg name="robot_ip" value="$(arg robot_ip)" />
    <arg name="report_type" value="$(arg report_type)" />
    <arg name="show_rviz" value="false" />

    <arg name="add_other_geometry" default="$(arg add_other_geometry)"/>
    <arg name="geometry_type" default="$(arg geometry_type)"/>
    <arg name="geometry_mass" default="$(arg geometry_mass)"/>
    <arg name="geometry_height" default="$(arg geometry_height)"/>
    <arg name="geometry_radius" default="$(arg geometry_radius)"/>
    <arg name="geometry_length" default="$(arg geometry_length)"/>
    <arg name="geometry_width" default="$(arg geometry_width)"/>
    <arg name="geometry_mesh_filename" default="$(arg geometry_mesh_filename)"/>
    <arg name="geometry_mesh_origin_xyz" default="$(arg geometry_mesh_origin_xyz)"/>
    <arg name="geometry_mesh_origin_rpy" default="$(arg geometry_mesh_origin_rpy)"/>
    <arg name="geometry_mesh_tcp_xyz" default="$(arg geometry_mesh_tcp_xyz)"/>
    <arg name="geometry_mesh_tcp_rpy" default="$(arg geometry_mesh_tcp_rpy)"/>
  </include>

  
  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="initial_reset" value="true"/>
  </include>     

  
  <arg name="markerId"        default="555"/>      
  <arg name="markerSize"      default="0.07"/>    
  <arg name="eye"             default="left"/>
  <arg name="marker_frame"    default="camera_marker"/> 
  <arg name="ref_frame"       default="camera_color_frame"/>  
  <arg name="corner_refinement" default="LINES" />     
  <arg name="camera_frame" default="camera_color_frame" />
  <arg name="camera_image_topic" default="/camera/color/image_raw" />
  <arg name="camera_info_topic" default="/camera/color/camera_info" /> 

  <node pkg="aruco_ros" type="single" name="aruco_single">
    <remap to="$(arg camera_info_topic)" from="/camera_info" />
    <remap to="$(arg camera_image_topic)" from="/image" />
    <param name="image_is_rectified" value="true"/>
    <param name="marker_size"        value="$(arg markerSize)"/>
    <param name="marker_id"          value="$(arg markerId)"/>
    <param name="reference_frame"    value="$(arg ref_frame)"/>   
    <param name="camera_frame"       value="$(arg camera_frame)"/>
    <param name="marker_frame"       value="$(arg marker_frame)" />
    
  </node>

  <arg name="namespace_prefix" default="xarm_eoh_calib"/>
  
  <include file="$(find easy_handeye)/launch/calibrate.launch">
    <arg name="eye_on_hand" value="true"/>
    <arg name="namespace_prefix" value="$(arg namespace_prefix)"/>
    <arg name="move_group" value="xarm$(arg dof)"  doc="the name of move_group for the automatic robot motion with MoveIt!" />
    <arg name="freehand_robot_movement" value="false"/>

    
    <arg name="robot_base_frame" value="link_base"/>
    <arg name="robot_effector_frame" value="link_eef"/>

    
    <arg name="tracking_base_frame" value="$(arg camera_frame)"/>
    <arg name="tracking_marker_frame" value="$(arg marker_frame)"/>

    <arg name="robot_velocity_scaling" value="0.5" />
    <arg name="robot_acceleration_scaling" value="0.2" />
  </include>

</launch>

penglongxiang, 2021-09-26:
Hi @Aki92-Xavier, we are testing the camera calibration support, but this part is actually not ready for release yet, we uploaded it by mistake. Since it's still under development, we are not sure if this can give the correct calibration.

Aki92-Xavier, 2021-09-26:
Thank you for your quick reply.
I'm looking forward to see the camera calibration for xarm.

closed 2021-09-26 by Aki92-Xavier

jzx-gooner, 2021-12-14:
> Hi @Aki92-Xavier, we are testing the camera calibration support, but this part is actually not ready for release yet, we uploaded it by mistake. Since it's still under development, we are not sure if this can give the correct calibration.

now,can we use the camera calibration correctly? and can we simulation the vision detection in gazebo,is it only work in the real arm? Thank you very much

penglongxiang, 2021-12-14:
hi @jzx-gooner , yes it is ready now for realsense D435i with real xArm. Please refer to https://github.com/xArm-Developer/xarm_ros#7-xarm-vision

jzx-gooner, 2021-12-15:
> hi @jzx-gooner , yes it is ready now for realsense D435i with real xArm. Please refer to https://github.com/xArm-Developer/xarm_ros#7-xarm-vision

cool,i will try !

goyalvarun02, 2022-08-12:
> hi @jzx-gooner , yes it is ready now for realsense D435i with real xArm. Please refer to https://github.com/xArm-Developer/xarm_ros#7-xarm-vision

1. Eye-In-hand case
I was trying eye on hand calibration.  I have kept the Aruco marker on table and camera facing to Aruco marker . I was able to do check starting pose somehow . Thereafter clicked on next pose, plan and execute then End effector (with Gripper) moves to the position from where Aruco marker is not visible completely (only partially visible) . Then I try to click on taking sample and it failed  (as expected)  .

Then I tried to skip the sample where Aruco marker is not visible completely and then I lead to taking samples around 8-10. Then Compute and save the calibration but that was not precise. Is there any particular number of samples must be for precise coordinate?

2. Eye-on-Base
I have tried calibration multiple times changing the position of camera and robot . I am observing the coordinates of object returning from camera are not accurate,  it is +/- X/Y . So, gripper not able to go to the exact position of object although it missed the object. If possible, can you advise anything for this as well. 

May I know in which scenario “pre-empted” can occur? And reason of its occurrence while calibrating or executing using move it?
  
Thanks

penglongxiang, 2022-08-22:
hi @goyalvarun02,

1. For my own experience, if the marker moves out of vision, just skip this sample and click next. If after one cycle there is not enough valid sample points, then go with "check starting point" and do the whole process again until there are at least 15 valid samples.

2. Sorry I have no experience of Eye-On-Base calibration, you may consult "`easy-handeye`" developers/maintainers for advice. Or check if there is better calibration method other than this.  

3. "Preempted" is usually caused by real robot failing to follow the command trajectory during execution of a successful plan. Please check if there is any **Error Code** reported from xArm controller side.