xArm-Developer/xarm_ros #88: Robot joints going to 0 when launching Gazebo in ROS melodic

MarcFacerias, 2021-11-05:
When executing the examples with xarm7 I get the error "No p gain specified for PID.  Namespace: /xarm/gazebo_ros_control/pid_gains/joint2", which can be solved by uncommenting certain code in the YAML files, however, whenever I do that all the robot joints go to 0 and the simulator is unusable, how should I address it?

penglongxiang, 2021-11-08:
Hi @MarcFacerias, the error "No P gain specified" actually does not affect the simulation under position control. If you must use the PID gains, maybe tuning has to be performed to make stable simulations in Gazebo.

MarcFacerias, 2021-11-08:
Hello, @penglongxiang thank you for your response, my actual problem is that whenever I try to move the robot and locate it in another place within the simulation the model collapses, I thought the reason was some weird interaction between the PID controllers and colliding with an obstacle, but after your message, I understand that shouldn't cause that issue. You can find attached an image with its behaviour. 

Thanks in advance
[image]
e