xArm-Developer/xarm_ros #89: Issue about Motion Planner and calling of current, velocity and acceleration

Lu1993mun, 2021-11-26:
Hello, I am a user of Xarm6 from Memorial Unversity of Newfoundland. Currently, we are developing teleoperation with Xarm6 and we met the following problem. I am grateful if you can provide any help.
First, we tried Moveit! to do the motion planning task given a desired pose in Cartesian space and the result was not satisfied.  With neither KDL nor Trac-IK solver, the planner still have a nonnegligible chance to fail. So is there any bulid- in planner in the robot and is there any respective API to call?
Second, we also need information of velocity, acceleration and current of each joint for some calculations.  Could you please tell me whether these data is available and which API should I call.
Thanks

Lu1993mun, 2021-11-30:
@penglongxiang  @vimior

penglongxiang, 2021-12-02:
Hi, currently joint accelerations and jerks are not available in real-time updated manner. The single number returned stands for the maximum value applicable for all joints. We will do assessment about the real-time report possibility, for now you can try differentiating the velocity, maybe add some filtering.

Lu1993mun, 2021-12-06:
Hello, Thanks for your reply. 
Recently, when I fix Xarm6 at a certain pose and use the following code to get the torque of the motor:

		_for (int i = 0; i <20; i++) {
			print_nvect("* joints_torque:", arm->joints_torque, 7);
			print_nvect("* realtime_joint_speeds:", arm->realtime_joint_speeds, 7);
			sleep_milliseconds(500);
		}_
I found the values I got is not stable and I added the screenshot as below. I guess this is due to the sleep_milliseconds() function.  I just use it as a sleep() function to pause the program for a few seconds. However, it also does something to  the robot controller/ joint motor.  Is my understanding correct or there is other reasons? 
Since I did not find any  explaination for sleep_milliseconds(), could you plz explain what it does? Thanks

[image]
@penglongxiang

penglongxiang, 2021-12-08:
Hi, 
sleep_milliseconds() is just a system sleep function, it has nothing to do with robot controller.
As for the torque values, did you add any additional payload to the arm? Could you please share the configuration of xArm6 (joint angles), mounting and payload settings, we will test accordingly, thank you.

Lu1993mun, 2021-12-10:
Hi,
Nothing is attached on the arm. Here below is the code I run:
```
#include "xarm/wrapper/xarm_api.h"

int main(int argc, char** argv) {

	std::string port="192.168.1.245";

	XArmAPI* arm = new XArmAPI(port);
	sleep_milliseconds(500);
	if (arm->error_code != 0) arm->clean_error();
	if (arm->warn_code != 0) arm->clean_warn();
	arm->motion_enable(true);
	arm->set_mode(0);
	arm->set_state(0);
	sleep_milliseconds(500);

	printf("=========================================\n");

	int ret;
	arm->reset(true);                                       /* Make the robot returning to the origin*/
	system("pause");
	fp32 poses[6][6] = {
		{300, 0, 200, 180, 0, 0},
		{300, 200, 200, 180, 0, 0},
		{500, 200, 200, 180, 0, 0},
		{500, -200, 200, 180, 0, 0},
		{300, -200, 200, 180, 0, 0},
		{300, 0, 200, 180, 0, 0},
	};
	for (int i = 0; i < 2; i++) {
		ret = arm->set_position(poses[i], true);            /* Parameter true: whether to wait for the arm to complete, default is false */
		printf("set_position, ret=%d\n", ret);
		}
	for (int i = 0; i < 50; i++) {
		print_nvect("* joints_torque:", arm->joints_torque, 7);
		print_nvect("* realtime_joint_speeds:", arm->realtime_joint_speeds, 7);
		sleep_milliseconds(500);                               
	}
	system("pause");
	arm->reset(true);
	return 0;
}
```
And here is the results:
[](url
[image]
)
The status of xarm6:
[image]
@penglongxiang

penglongxiang, 2021-12-13:
hi, it seems you are just using the C++ SDK instead of ROS interface. I have tried the same code on our robot and the output was: 

```
SDK_VERSION: 1.8.5
Tcp control connection successful
FIRMWARE_VERSION: v1.8.5, PROTOCOL: V1, DETAIL: 6,6,XI1300,DC1300,v1.8.5
Tcp Report Rich connection successful
report_data_size: 494, size_is_not_confirm: 0
=========================================
set_position, ret=0
set_position, ret=0
* joints_torque:-0.000 -12.665 -9.905 -0.094 -0.999 -0.000 0.000 
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000 
* joints_torque:-0.000 -12.665 -9.906 -0.094 -0.999 -0.000 0.000 
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000 
* joints_torque:-0.000 -12.665 -9.907 -0.094 -0.999 -0.000 0.000 
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000 
* joints_torque:-0.002 -12.665 -9.907 -0.094 -0.999 0.000 0.000 
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000 
* joints_torque:0.000 -12.665 -9.907 -0.094 -0.999 0.000 0.000 
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000 
* joints_torque:0.000 -12.665 -9.907 -0.094 -0.999 0.000 0.000 
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000 
* joints_torque:-0.000 -12.665 -9.907 -0.094 -0.999 0.000 0.000 
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000 
* joints_torque:-0.000 -12.665 -9.907 -0.094 -0.999 0.000 0.000 
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000 
* joints_torque:-0.000 -12.665 -9.907 -0.094 -0.999 0.000 0.000 
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000 
* joints_torque:-0.000 -12.665 -9.907 -0.094 -0.999 0.000 0.000 
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000 
* joints_torque:-0.000 -12.665 -9.907 -0.094 -0.999 0.000 0.000 
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000 
* joints_torque:-0.000 -12.665 -9.907 -0.094 -0.999 0.000 0.000 
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000 
* joints_torque:-0.000 -12.665 -9.907 -0.094 -0.999 0.000 0.000 
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000 
```
Which seems normal. Maybe it was a version incompatibility problem, please try upgrading the xArm controller firmware and your locally installed xarm C++ SDK to the latest version and try again. Thanks.

Lu1993mun, 2021-12-13:
Hello,
I update the controller firmware and now it is in the same version of SDK.  However, the output is still not as yours. Is there any other reason which might lead this?
```
Microsoft Windows [版本 10.0.19042.1348]
(c) Microsoft Corporation。保留所有权利。

C:\Users\DPBG>cd C:\C++\xArm-CPLUS-SDK-master\visual_studio\x64\Debug\

C:\C++\xArm-CPLUS-SDK-master\visual_studio\x64\Debug>9998-Sensor_data.exe
SDK_VERSION: 1.8.5
Tcp control connection successful
FIRMWARE_VERSION: v1.8.5, PROTOCOL: V1, DETAIL: 6,6,XI1203,AC1300,v1.8.5
Tcp Report Rich connection successful
report_data_size: 494, size_is_not_confirm: 0
=========================================
请按任意键继续. . .
set_position, ret=0
set_position, ret=0
* joints_torque:3.136 -9.924 -12.623 -0.835 0.661 1.636 0.000
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000
* joints_torque:2.348 -10.684 -11.644 -0.831 0.475 1.525 0.000
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000
* joints_torque:1.228 -11.838 -10.243 -0.860 0.135 1.287 0.000
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000
* joints_torque:0.457 -12.591 -9.423 -0.876 -0.018 1.102 0.000
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000
* joints_torque:-0.414 -13.760 -9.029 -0.898 -0.408 0.937 0.000
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000
* joints_torque:-0.553 -14.779 -9.096 -0.907 -0.756 0.659 0.000
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000
* joints_torque:-0.469 -15.183 -9.180 -0.905 -0.985 0.515 0.000
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000
* joints_torque:-0.322 -15.436 -9.361 -0.918 -1.256 0.303 0.000
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000
* joints_torque:-0.189 -15.521 -9.486 -0.916 -1.371 0.144 0.000
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000
* joints_torque:-0.052 -15.463 -9.653 -0.932 -1.536 0.000 0.000
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000
* joints_torque:-0.002 -15.393 -9.788 -0.931 -1.599 -0.125 0.000
* realtime_joint_speeds:0.000 0.000 0.000 0.000 0.000 0.000 0.000

```

penglongxiang, 2021-12-14:
Hi, please keep outputting these torque for a longer period to see if the values can become stable eventually. Meanwhile, please share the latest result and refer this issue to support@ufactory.cc, additional information would be the Serial Number of your xArm. Our support team will further assist on this issue. Thank you.

Lu1993mun, 2021-12-14:
Hi, thanks for your reply and patience. I suddenly found there is seven outputs in your result. I guess you run my code on xarm7 and mine is xarm6.  Could you plz test my code on xarm6 if my guess is right? 
By the way should I use `print_nvect("* joints_torque:", arm->joints_torque, 6);`  instead of `print_nvect("* joints_torque:", arm->joints_torque, 7);` in order to print the torque? or parameter 7 does not matter.
If your result is stable, I will contact support@ufactory.cc later. Thanks

penglongxiang, 2021-12-15:
I also run the code on xArm6, there are always 7 output values but only the first 6 are useful for xArm6.