xArm-Developer/xarm_ros #91: ask some questions

jzx-gooner, 2021-12-14:
1. `roslaunch xarm_description xarm7_rviz_display.launch`  the modle is not good ,and the error in rviz  `No tf data.  Actual error: Fixed Frame [world] does not exist`  and the error `No transform from [link1] to [world]`in robotmodel
2. `$ roslaunch xarm_gazebo xarm7_beside_table.launch [run_demo:=true] [add_gripper:=true] [add_vacuum_gripper:=true] ` the model is good ,however it do not run the trajectory move.
3. `rosrun xarm_gazebo color_recognition.py` modify to python2 and seems it do not work to grasp.
Thank you very much!

penglongxiang, 2021-12-14:
1. This launch script behaves normally on our side. Please make sure all dependencies have been properly installed. If issue still persists, please tell us your running environment and ROS version.
2. Try removing the brackets `[ ]` if you have added them in command line. Here, they supposed to mean "optional arguments", and please note you can not add the gripper and vaccum_gripper at the same time. 
3. This script is still under test, and not yet appeared in our ReadMe. Please do not use it yet with python2.

jzx-gooner, 2021-12-15:
Thank you for your help.the problem solved!

closed 2021-12-15 by jzx-gooner