xArm-Developer/xarm_ros #92: realMove_exec.launch: controller failed during execution (xarm6)

iskanderuse, 2021-12-14:
Hello, TrajectoryExecutionManager of Moveit  ( ROS, melodic, ubuntu 18.04, real robot xarm-6, xarm_ros (master)) throws such error:

`[ERROR][/move_group][TrajectoryExecutionManager::configure][1116]: Unable to identify any set of controllers that can actuate the specified joints: [ joint1 joint2 joint3 joint4 joint5 joint6 ]
[ERROR][/move_group][TrajectoryExecutionManager::configure][1130]: Known controllers and their joints:
`
`[planning_interface::MoveItErrorCode moveit::planning_interface::MoveGroupInterface::MoveGroupInterfaceImpl::execute][833]: ABORTED: Solution found but controller failed during execution
`
It seems that the error in the namespace...
Can you help me?

penglongxiang, 2021-12-15:
Please use `dpkg -l | grep ros-control` to check if `ros-<distro>-ros-control` and `ros-<distro>-ros-controllers` have been properly installed. If not, please follow the section 7 in http://wiki.ros.org/ros_control to install.

iskanderuse, 2021-12-15:
`ros-<distro>-ros-control` and `ros-<distro>-ros-controllers` have been installed

**dpkg -l | grep ros-control**
`ii  ros-melodic-gazebo-ros-control                                     2.8.7-1bionic.20210921.213902       amd64        gazebo_ros_control
ii  ros-melodic-moveit-ros-control-interface                           1.0.8-1bionic.20211005.004851       amd64        ros_control controller manager interface for MoveIt!
ii  ros-melodic-moveit-ros-control-interface-dbgsym                    1.0.8-1bionic.20211005.004851       amd64        debug symbols for ros-melodic-moveit-ros-control-interface
ii  ros-melodic-ros-control                                            0.18.4-1bionic.20210921.220930      amd64        A set of packages that include controller interfaces, controller managers, transmissions and hardware_interfaces.
ii  ros-melodic-ros-control-boilerplate                                0.5.2-1bionic.20210921.220916       amd64        Simple simulation interface and template for setting up a hardware interface for ros_control
ii  ros-melodic-ros-controllers                                        0.17.2-1bionic.20211005.002411      amd64        Library of ros controllers

penglongxiang, 2021-12-15:
Hi @iskanderuse , it was due to a bug of my recent update. This issue is already fixed, please checkout the latest version and run it again. Sorry for the inconvenience.

iskanderuse, 2021-12-15:
Thank You, now it is working

closed 2021-12-15 by iskanderuse

AtariaLee, 2022-11-09:
I have the same problem!how could i solve it.