xArm-Developer/xarm_ros #93: Usb-camera cable mounting on x-arm

iskanderuse, 2021-12-17:
I hope this question is relevant here. It worries me a lot, because I have already ruined 2 expensive cables for the camera ...
Is there a way to mount a stereo camera so that the camera cable does not twist around the robot when the joints move, and the USB cable connector does not pull out of the camera?
Thank you in advance for your advice on how best to fix the camera cable on the x-arm.

BryanStuurman, 2021-12-20:
We've had issues like this in the past as well and the best way to mitigate it is strain relieving the cable across the two joints. Referring to the attached picture, you can see the usb cable is strain relieved at the realsense (it's screwed in, good on intel for providing the holes) and at the "forearm" using the zip ties.
[image]

after setting this up I like to kick the robot into manual mode ("zero G mode" ) and articulate it through it's range of motion by hand and make sure that the cable doesnt become stressed - if it does just add more cable, or more zip ties.

scottrpaterson, 2022-05-16:
On this topic, the original marketing material for the xARM showed a camera built into the gripper. Has anyone tried to split the RS485 connector in two and pass the camera feed through the existing cables to the control box?

BryanStuurman, 2022-05-19:
Scott, I just had the opportunity to check continuity from the end effector IO connector to the robot control connector and there is no continuity from "User 485" to any of the control pins - I'm guessing they have a dedicated repeater structure at the end effector to give the User 485 functionality.

TonyLeheng, 2022-05-19:
Hey guys, cameras need high-speed ports like USB or ethernet, which requires high transmission quality of the cables inside.  And the current version robot doesn't support this which means even if you find a way to connect the camera through internal cables, the signal quality will be really bad and unusable.
Moreover, the tool RS485 cable is not physically connected to the control box, there is a transmission module inside, since we use higher baudrate with special protocol internally, but Modbus usually runs at 115200bps.

TonyLeheng, 2022-05-20:
There is no termination resistor between AB wires. In our design, the resistor is on the tool side, so it is also easier to adjust and match different head tools. xArm doesn't have AC coupling at all, it is a standard RS485 signal cable with a higher baud rate.