xArm-Developer/xarm_ros #97: link5 / end_tool model inconsistency

BryanStuurman, 2022-02-03:
Introduced in commit 29bcd68, end_tool.stl doesn't line up with link5.stl (it is "down" in the world) and interferes with my custom end effector collision model.

I can work around this by replacing the reference to end_tool with link5, however it raises a much more serious question: which is the correct model? 

my ask: Can the Ufactory team review the models and correct the discrepancy?

penglongxiang, 2022-02-10:
Hi @BryanStuurman, I have done the checking and found collision model is shorter in hight (Z direction) by 1 mm. Please use the visual model as the correct one, since collision model is a simplified version for faster collision check. We will try to update the collision model in future update, thanks for pointing out the discrepancy.

penglongxiang, 2022-02-10:
Hi, I am attaching the updated "end_tool.stl" here for your test. Please replace and check if this aligns with other links for collision check.
[end_tool.STL.zip](https://github.com/xArm-Developer/xarm_ros/files/8038956/end_tool.STL.zip)